robotology/mesh-support #1502: Neck joint cable broken

GiulioRomualdi, 2023-02-14:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While testing the walking controller the neck cable got broken

### Detailed context

@sgiraz, @AntonioConsilvio, and @GiacomoVassallo13 blocked the neck with a plastic tie

sgiraz, 2023-02-15:
[image]

pattacini, 2023-02-15:
The neck is back operational ✅
Thanks to @fbiggi @mpinaffo @sgiraz @mfussi66 @salvi-mattia.

We spotted that you were not using the latest set of PID gains, which goes in the direction of weighing robustness more than tracking performance.

We left the new gains in your fork (both in `src` for version control and in the installed dir).
Also, we recalibrated the AEA sensors, so you'll find these new values too.

@mfussi66 can you drop a pointer to the correct set?

sgiraz, 2023-02-15:
Hi @mfussi66,

This morning the head has been fixed. Could you post the video here that proves the correct behavior when running the `ergoCub` head?

mfussi66, 2023-02-15:
The most recent pid values to control roll and pitch of the head can be found here: https://github.com/robotology/robots-configuration/blob/master/ergoCubSN000/hardware/motorControl/head-eb20-j0_1-mc.xml#L50-L52

pattacini, 2023-02-15:
> The most recent pid values to control roll and pitch of the head can be found here: https://github.com/robotology/robots-configuration/blob/master/ergoCubSN000/hardware/motorControl/head-eb20-j0_1-mc.xml#L50-L52

`devel` branch updated as well via:
- https://github.com/robotology/robots-configuration/commit/9ff9d9087682b222fa569310b9421f336e322a52

closed 2023-02-16 by pattacini