robotology/mesh-support #1503: Ankle roll motor moving

lrapetti, 2023-02-16:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The ankle roll motor seems to move when pushed.

### Detailed context

https://user-images.githubusercontent.com/35487806/219408178-b4b12971-44ca-48aa-b68b-5648e65e9b12.mp4

sgiraz, 2023-02-16:
After tightening the screws the problem should be solved.
Maybe some thread-locker may help.

cc @maggia80 @AntonioConsilvio @Fabrizio69 @fbiggi

DanielePucci, 2023-02-21:
This keeps happening unfortunately 

| https://user-images.githubusercontent.com/5045846/220378031-575e57cc-2f89-494c-85d1-03fff4730ccd.mp4 | https://user-images.githubusercontent.com/5045846/220378839-cf0b0268-2aee-4265-b2a4-2de7f36307be.mp4 |
|:--:|:--:|

lrapetti, 2023-02-22:
Yesterday @AntonioConsilvio tighten the motor screws again. However it would be good to implement a solution in order to avoid it happening again

AntonioConsilvio, 2023-02-22:
Hi @lrapetti , @GiacomoVassallo13 and me put thread-locker on the reachable screws.

So when you will use the robot, please send us feedback if it is enough, or we have to do it for each screw and then we have to disassemble the feet.

cc @sgiraz @maggia80 @Fabrizio69

sgiraz, 2023-02-23:
The `thread-locker` seems to do the job! 

Closing!

closed 2023-02-23 by sgiraz