robotology/mesh-support #1504: Remove forearms for mounting iRonCub pieces

HosameldinMohamed, 2023-02-16:
### Device name 🤖

iCubGenova09

### Request/Failure description

We would like to remove the forearms of iCub3 to mound 3D-printed pieces to perform some tests with iRonCub.

CC @vpunithreddy @gabrielenava 

### Detailed context

We know how to remove the forearms mechanically, but we might need support to remove the electronic parts!

### How does it affect you?

We need to perform some tests for iRonCub.

HosameldinMohamed, 2023-02-16:
@AntonioConsilvio did it and showed us the necessary cables to disconnect in order to remove the cables!

I'll write down here the steps needed to do that.

HosameldinMohamed, 2023-02-16:
## Right arm

- Remove the upper arm's cover.
- Remove the skin cable 

[image]

- Disconnect the 3 connectors shown below:

[image]

1. Power for the forearm.
2. Ethernet cable going to the forearm boards from EMS board `EB4` (E4).
3. Ethernet cable coming from the forearm boards to Waist EMS board `EB5` (`E29`).

- Remove the small 3D-printed white plate holding the cable:
[image]

- Unmount the screws of the forearm (details later!)

## Left arm

- Remove the upper arm's cover.
- Remove the skin cable 
- Disconnect the 3 connectors shown below:
[image]

1. Power for the forearm.
2. Ethernet cable going to the forearm boards from EMS board `EB2` (E2).
3. Ethernet cable coming from the forearm boards to Right arm EMS board `EB3` (`E26`).

- Remove the small 3D-printed white plate holding the cable.
- Unmount the screws of the forearm (details later!)

HosameldinMohamed, 2023-02-16:
Then we need to bypass the ethernet loop.

We need jumper cables to complete the loop because the types of connectors are different!

The diagram shows where to place the jumpers, from https://icub-tech-iit.github.io/documentation/icub_wiring/icub3_x/#icub-3x-wiring

[image]

We don't have these jumper cables so we'll open another ticket asking to make them.

sgiraz, 2023-02-16:
Hi @HosameldinMohamed,

I guess in the next few days you would re-mount the forearms.
Keep us updated about the progress of these activities (as you're already doing), please.

HosameldinMohamed, 2023-02-17:
> Hi @HosameldinMohamed,
> 
> I guess in the next few days you would re-mount the forearms. Keep us updated about the progress of these activities (as you're already doing), please.

Hi @sgiraz! Yes, the test will probably take a few days. In addition to #1506, we are also waiting for some metal parts to arrive to mount them on the robot.

I would keep the robot in this state until completing our tests, to avoid remounting/unmounting the forearms many times.

In the meantime, if anyone needs to use the robot, We'll put mount the forearms back! I wrote it in the Teams channel..

Do you prefer keeping this issue for the updates, or using the Teams channel for that?

sgiraz, 2023-02-18:
> Do you prefer keeping this issue for the updates, or using the Teams channel for that?

I suggest keeping track here of all physical (mechanical/electrical) changes made to the `iCub3` during your tests.

cc @AntonioConsilvio @GiacomoVassallo13

AntonioConsilvio, 2023-02-20:
Hi @HosameldinMohamed, the only change made is the one in this [issue](https://github.com/robotology/icub-tech-support/issues/1506#issuecomment-1436743781)

cc @sgiraz

AntonioConsilvio, 2023-02-21:
Hi @HosameldinMohamed, when you think you have all the information about this activity, you can close the issue.

In any case, I am always available for any questions in the future as well!

cc @sgiraz

HosameldinMohamed, 2023-02-21:
> Hi @HosameldinMohamed, when you think you have all the information about this activity, you can close the issue.

Hi @AntonioConsilvio, we have only the mechanical part left to write it down (if you think the electronic part is OK!). I expect to do it by today!

> In any case, I am always available for any questions in the future as well!

Thanks!

sgiraz, 2023-02-23:
Forearms with electronic parts have been removed.
You should now be able to unmount and remount the forearms independently.

Well done @AntonioConsilvio! 🚀

closed 2023-02-23 by sgiraz