robotology/mesh-support #1505: Robot not fixed properly on the pole

lrapetti, 2023-02-16:
### Device name 🤖

ergoCub

### Request/Failure description

After placing the robot on the pole we noticed the robot pelvis was not properly fixed (see video below).

### Detailed context

As suggested by @maggia80 it should be enough to tight the screw on the pole, we should verify if it gets properly fixed

https://user-images.githubusercontent.com/35487806/219438321-7d4d76f2-354f-4ee5-9821-e918b7c0ae76.mp4

cc @mebbaid @GiulioRomualdi

Gandoo, 2023-02-17:
ciao al momento è stata accorciata la curvatura e il problema è quasi risolto, in aggiunta ho incollato uno spessore.
In attesa di una nuova staffa di ancoraggio

[image]

cc @maggia80 @sgiraz @AntonioConsilvio @DanielePucci @lrapetti

sgiraz, 2023-02-17:
Well done @Gandoo! 🚀

lrapetti, 2023-02-17:
At the beginning the fix seems to work out, but after a while it was moving again

https://user-images.githubusercontent.com/35487806/219676880-7e1b516f-33f1-4393-9eb0-023f2f4118f8.mp4

sgiraz, 2023-02-18:
Hi @lrapetti,

Thank you for your feedback. Next sprint, we will probably deliver an improved lock mechanism.

lrapetti, 2023-02-20:
> Hi @lrapetti,
> 
> Thank you for your feedback. Next sprint, we will probably deliver an improved lock mechanism.

Related to the pole we also want to ask for having the possibility to level the pole (e.g. by adjusting the height of the wheels), since it can band under the weight of the robot itself:

[image]

Elio75, 2023-02-22:
> At the beginning the fix seems to work out, but after a while it was moving again

To remove the backlash of the robot in the indicated area, we removed 0.4 mm on each side on the centering supports. It seems to have worked because the robot remains quite stable and does not move.

[image]

CC @maggia80 @Fabrizio69

AntonioConsilvio, 2023-02-22:
@lrapetti the backlash should be gone, however when you will use the robot, send us feedback about it.

cc @sgiraz @Elio75

lrapetti, 2023-02-23:
> @lrapetti the backlash should be gone, however when you will use the robot, send us feedback about it.
> 
> cc @sgiraz @Elio75

We tested it just after the modification of the support and the robot was stable on the pole. I think we can close this, and we will inform you if we encounter the same issue next time the robot is on the pole.

Just as a reminder, if possible take in consideration https://github.com/robotology/icub-tech-support/issues/1505#issuecomment-1437018373 for the future iterations on robot poles

closed 2023-02-23 by lrapetti