robotology/mesh-support #1506: Build jumper cables to bypass ethernet connections when removing forearms

HosameldinMohamed, 2023-02-16:
### Device name 🤖

iCubGenova09

### Request/Failure description

We removed the forearms in #1504 to perform a test for iRonCub. We need to have an ethernet cable to bypass the forearms for completing the ethernet loop.

The connectors are of different types so we need small jumper cables to have compatibility with the connectors.

### Detailed context

See https://github.com/robotology/icub-tech-support/issues/1504#issuecomment-1433484053

## Right arm

[image]

## Left arm

[image]

AntonioConsilvio, 2023-02-20:
Hi @HosameldinMohamed, I changed this connector (6 pins)

[image]

With a new one (5 pins):

[image]

So you can connect it on the board, and also on the male connector of the fore arm:

[image]

If you have to connect the cable with the male connector of the fore arm, be careful because one PIN is empty and you have to be sure that you are connecting pin to pin.

When you have tested the robot, give us feedback.

cc @sgiraz

HosameldinMohamed, 2023-02-20:
Hi @AntonioConsilvio 
We tested the robot.. it can start with no issues!

Thanks!

closed 2023-02-20 by AntonioConsilvio