robotology/mesh-support #1508: iCubGenova11 - Left elbow is mechanically locked

xEnVrE, 2023-02-20:
### Device name 🤖

iCubGenova11

### Request/Failure description

While doing experiments with iCubGenova11 for the ergoCub demos, @Woolfrey notified us that the left elbow suddenly stopped moving.

At the moment, the elbow motion seems blocked by something, a cable?, as can be seen in the following video:

https://user-images.githubusercontent.com/6014499/220168300-1d7bba22-18d1-4d97-9a74-ff5f390990d9.mp4

### Detailed context

@Woolfrey has a video showing what the robot was actually doing when the block came into place. I will upload it here as soon as it is ready.

### How does it affect you?

We cannot test one of the demo for the ergoCub demos day.

AntonioConsilvio, 2023-02-21:
Hi @xEnVrE, @GiacomoVassallo13 and I tightened the tensioners of the steel cables. It is working now. (also with `yarprobotinterface`)

https://user-images.githubusercontent.com/114915464/220305454-9b7432e7-1cd7-4dd5-862d-827a98b2ea6b.mp4

cc @sgiraz

closed 2023-02-21 by AntonioConsilvio

xEnVrE, 2023-02-21:
Thank you @AntonioConsilvio @GiacomoVassallo13