robotology/mesh-support #1511: R1 Broken led display

randaz81, 2023-02-23:
### Device name 🤖

R1SN001 Erzelli

### Request/Failure description

The led matrix does not turn on. In few occasions it turn on, showing random pixels.
I suspect a broker flat cable connector on the fpga board because touching it with my finger, the board some leds turns on. The flash cable was already replaced unsuccessfully.

The realsense camera turns offs when the robot head moves.  (Broker usb cable insider the head?)

We have a demo with important people on March 1st.

### Detailed context

..

sgiraz, 2023-02-23:
Activity planned for tomorrow morning!

cc @AntonioConsilvio @Fabrizio69 @fbiggi @randaz81

AntonioConsilvio, 2023-02-24:
Hi, @Gandoo @fbiggi and I replaced the board that powers the front screen (part number: 10830.A). 
Then we also replaced the Real Sense cable.

Then, when we tested the robot, the neck motors stopped working. 
I replaced the MC4-Plus that controls these motors with a new one. The motors started working, but another problem appeared.

The motor that controls the rotation of the neck probably has the pinscrew loose. Therefore, we should schedule an intervention.

cc @sgiraz @Fabrizio69 @randaz81

sgiraz, 2023-02-24:
Cc @fbiggi

fbiggi, 2023-02-27:
cc @Gandoo

sgiraz, 2023-03-22:
/remind activity planned for Wednesday 29 March

sgiraz, 2023-03-22:
/remind Prepare for the activity 28 March 9:00

sgiraz, 2023-03-22:
/remind Activity scheduled for tomorrow 

@set-reminder 28 March 9:00

sgiraz, 2023-03-22:
Activity scheduled for tomorrow!

@set-reminder 28 March 9:00

sgiraz, 2023-03-22:
/remind 28 March 9:00 Activity scheduled for tomorrow!

AntonioConsilvio, 2023-03-29:
This morning, @fbiggi and I solved the neck problem by tightening the pinscrew in the neck.

The robot is now working properly:

https://user-images.githubusercontent.com/114915464/228555386-a2c161da-2d34-4a3c-8bfb-d77217dd412d.mp4

cc @sgiraz @Fabrizio69

sgiraz, 2023-03-29:
Well done guys! 🎉

closed 2023-03-29 by sgiraz