robotology/mesh-support #1512: iCubGenova11 - Problem with r_thumb_proximal

fedeceola, 2023-02-23:
### Device name 🤖

iCubGenova11

### Request/Failure description

During an experiment, the `r_thumb_proximal` stopped working. After this, when the `yarprobotinterface` is launched, the calibration of the right hand is not completed. 

### Detailed context

-

xEnVrE, 2023-02-23:
The robot is at the moment used for both preparing the incoming ergoCub demos and for an IROS 2023 conference submission

Would it be possible to check this tomorrow morning? @sgiraz @AntonioConsilvio 

Thank you

sgiraz, 2023-02-23:
Until 1 p.m.  @AntonioConsilvio will not be available.
If @GiacomoVassallo13 will be available, he may help you in case of mechanical failure.
I'll let you know in the morning!

xEnVrE, 2023-02-24:
Thanks @sgiraz

sgiraz, 2023-02-24:
@GiacomoVassallo13 is going to help you to fix the thumb this morning!

GiacomoVassallo13, 2023-02-24:
Hi, the problem was the broken tendon. The tendon has be cutted because the dowel pin of pulley of joint was broke. Therefore I proceeded to replace the assembly and rewire the joint. The problem is solved, I attend a feedback.

cc: @xEnVrE @sgiraz

fedeceola, 2023-02-24:
Thanks @GiacomoVassallo13, I will make some tests in a few minutes.

fedeceola, 2023-02-24:
I just tried to run the `yarprobotinterface` and the right hand calibration still is not completed.

AntonioConsilvio, 2023-03-02:
The steel tendon got damaged again. It has been replaced by @GiacomoVassallo13!
The finger has been tested and seems work. 

cc @sgiraz

sgiraz, 2023-03-02:
Ok, super @AntonioConsilvio and @GiacomoVassallo13! 🚀

Closing!

closed 2023-03-02 by sgiraz