robotology/mesh-support #1514: iCubGenova11 - Problem with r_shoulder_yaw.

fedeceola, 2023-02-24:
### Device name 🤖

iCubGenova11

### Request/Failure description

### Device name 🤖

iCubGenova11

### Request/Failure description

During an experiment, the `r_shoulder_yaw` went on hardware fault, although it was in a feasible position. After this, when the `yarprobotinterface` is launched, all the r_shoulder joints remain in idle. Moreover, with the robot turned off, it is not possible to manually move the `r_shoulder_yaw`. 

### Detailed context

### Detailed context

-

xEnVrE, 2023-02-27:
Hi @sgiraz,

I wanted to ask if it would be possible to check this robot today.

Thanks

sgiraz, 2023-02-27:
Hi @xEnVrE , @AntonioConsilvio is going to reach you this morning!

fedeceola, 2023-02-27:
Hi @sgiraz, @AntonioConsilvio, as @xEnVrE mentioned in #1512 we have a couple of deadlines for an iROS submission (on the 1st of March) and for the ErgoCub demos (6th of March). Do you think that iCub can be repaired before those days?

sgiraz, 2023-02-27:
Hi @fedeceola,

Unfortunately we're having an emergency with `ergoCub` which currently has an higher priority.
We'll do our best to address your issue asap.

cc @AntonioConsilvio @GiacomoVassallo13

xEnVrE, 2023-02-27:
Thanks for the update @sgiraz

fbiggi, 2023-02-28:
Eseguita la seguente lavorazione
scabalggio
Cambio cuscinetto 618/4 
pulizia + aggiunta grasso nuovo su wave e flex 
Ispezione statore + pulizia disco ottico
Rimontaggio + chiusura secondo specifica Nm 
Prova di movimento manuale **conforme** in attesa di test di movimento elettrico

[image]
[image]
@AntonioConsilvio @GiacomoVassallo13 @sgiraz @maggia80 @Fabrizio69

xEnVrE, 2023-02-28:
This evening the robot was back in the robot arena.

We tried running the `yarprobotinterface` but it seems that something is wrong encoders-side as the shoulder did not calibrate:

[image]

the error message is `MC: AEA encoder invalida data.`

cc @sgiraz @AntonioConsilvio @fedeceola

sgiraz, 2023-02-28:
Ok, it should be something ez to fix. @AntonioConsilvio will help you tomorrow (1st March) morning.

AntonioConsilvio, 2023-03-02:
Hi @xEnVrE @fedeceola, the problem should be in the cable that connects the yaw AEA to the EMS board (eb3) (cable name: R1S0).
The robot has been tested and it should work, however send a feedback to us when you will use the robot.

cc @sgiraz

sgiraz, 2023-03-02:
Well done @AntonioConsilvio, I think we can close this issue!
Feel free to re-open if necessary.

closed 2023-03-02 by sgiraz