robotology/mesh-support #1515: neck cable out of rail

lrapetti, 2023-02-27:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The neck cable moved out of its rail.

### Detailed context

While moving the robot in position controlled.

sgiraz, 2023-02-27:
@GiacomoVassallo13, @AntonioConsilvio, and I are trying to fix it asap! 

We are trying to figure out the possible reasons why this issue keeps happening on `ergoCubSN000`:
- loose tendons ❓
- screws not tightened correctly ❓
- 3x shim used used improperly ❓

cc @Fabrizio69 @maggia80 @pattacini @Nicogene

sgiraz, 2023-03-01:
Actually, the neck seems to be fixed!

Waiting for further feedback.