robotology/mesh-support #1516: Face expressions display flickers

xEnVrE, 2023-02-28:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While using the robot we had to fault it and we accidentally hit the head display.

After that, a band on the right part of the display was flickering and sometimes was completely blank.

[image]

After touching the display flat a bit, it came back normal but we noticed two columns of the display were not working:

[image]

### Detailed context

.

sgiraz, 2023-02-28:
cc @Nicogene @Fabrizio69 @maggia80

maggia80, 2023-03-01:
I guess it's an hw issue. Let's discuss whether to change it now or after the demo.

sgiraz, 2023-03-02:
We're going to replace the display with a new one asap.

cc @AntonioConsilvio

sgiraz, 2023-03-06:
Display replaced by @AntonioConsilvio!
We saw during the INAIL demo it worked as expected.

Closing!

closed 2023-03-06 by sgiraz