robotology/mesh-support #1517: Weird tracking behavior for r_shoulder_roll

xEnVrE, 2023-02-28:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While testing the joint control of the left arm, we noticed that the `r_shoulder_roll` joint was departing from time to time from the expected trajectory commanded by the user using `POSITION_DIRECT`.

Below you can see:
- the desired trajectory sent to the joint (orange) vs the actual joint configuration (blue)
[image]
- the PWM action (orange) and the motor current (blue)
[image]

As it can be seen, the joint is tracked well but at certain point it departs from the desired trajectory.

### Detailed context

.

sgiraz, 2023-02-28:
cc @pattacini @Nicogene

sgiraz, 2023-02-28:
Ok @xEnVrE, thanks for reporting!
Unfortunately, many demos are coming soon and we have a few critical issues to address before.

As soon as the demo on 6 March is over, we gonna check out the problem reported here.

lrapetti, 2023-03-02:
Today it happened that the motor encoder was not reading, and after touching the connector it was back working. Probably it is worth to have a look at the connector, maybe changing it

sgiraz, 2023-03-02:
> Today it happened that the motor encoder was not reading, and after touching the connector it was back working. Probably it is worth to have a look at the connector, maybe changing it

Hi @lrapetti any update on this?

lrapetti, 2023-03-02:
Today @AntonioConsilvio fixed the connector, and we no longer experienced problems. Feel free to close this issue.

sgiraz, 2023-03-02:
Ok, thanks! Closing.

closed 2023-03-02 by sgiraz