robotology/mesh-support #1518: Cannot control wrists in position direct

xEnVrE, 2023-03-01:
### Device name 🤖

ergoCubSN000

### Request/Failure description

We cannot control the wrists using position direct mode, while they work using standard position control.

Here it is a video showing the behavior from the `yarpmotorgui`:

https://user-images.githubusercontent.com/6014499/222116483-09cacea2-76e3-45fd-a4d3-d8c0c6e3c5cb.mp4

### Detailed context

.

xEnVrE, 2023-03-01:
cc @sgiraz @AntonioConsilvio @lrapetti @Woolfrey

pattacini, 2023-03-01:
Thanks for reporting @xEnVrE 👍🏻 

We will investigate the FW of the AMC, which is responsible for dealing with the control modes and for generating the current setpoints sent to the AMCBLDC boards.

### Side note
This is not an HW issue, thus this is not the proper repo, but given the looming demos, that's ok 👍🏻

xEnVrE, 2023-03-01:
Thanks heaps @pattacini 

> This is not an HW issue, thus this is not the proper repo, but given the looming demos, that's ok 👍🏻

Yep, sorry for that, I didn't know where to open it exactly :)

pattacini, 2023-03-01:
>Yep, sorry for that, I didn't know where to open it exactly :)

The following options are ok:
- `icub-firmware`
- `icub-main`

xEnVrE, 2023-03-01:
Thanks @pattacini 

Should I move the issue there or it is fine to leave it here?

ale-git, 2023-03-02:
Hi, I've added the direct mode to control modes available in the new wrist. I don't remember why I didn't include it before, but I can't see any reason right now. Here it is the .hex file. I'm at Cris, so if you need help to test it please call me.

 
[ems004-wrist-direct.zip](https://github.com/robotology/icub-tech-support/files/10869261/ems004-wrist-direct.zip)

pattacini, 2023-03-02:
Thanks @ale-git!

Can you also open up the PR for the code in the meanwhile?

pattacini, 2023-03-02:
Also, did you check the availability of other "stiff" control modes like the velocity control?

pattacini, 2023-03-02:
@maggia80 noted that you applied the modifications to the EMS, but we need to do the same with the AMC board.

sgiraz, 2023-03-02:
> @maggia80 noted that you applied the modifications to the EMS, but we need to do the same with the AMC board.

@ale-git, if needed I can help you with the integration of the same changes on the AMC board. Anyway, I guess that these changes have been applied in `JointSet.c;`, if it is the case, they will also be reflected on AMC board since the source is the same.

ale-git, 2023-03-02:
Sorry, I didn't know that the EMS was already been replaced by the AMC. I asked to Marco A. which AMC project I have to use, but if @sgiraz knows it I can push the update and he can build it since as he sait it affects only JointSet.c that is shared. @pattacini other stiff modes should work. The unsupported modes are openloop, current and torque because they don't have a straightforwad meaning - we should decide one before.

ale-git, 2023-03-02:
https://github.com/robotology/icub-firmware/pull/351

pattacini, 2023-03-02:
Thanks heaps @ale-git!

We can give it a quick try on the setup wrist we have in the mechatronic lab before flashing the patched FW to the robot (as the robot may be unavailable now).

GiulioRomualdi, 2023-03-02:
Hi @pattacini, if the flashing procedure is the same (via Firmware updater) we can do it as soon as you provide us with the firmware. (It would be really helpful if you give us also the IP of the boards we have to flash so we are sure we do not make any mistakes)

marcoaccame, 2023-03-02:
surely @sgiraz knows but just as reminder:
- by default the binary is for the left wrist
- use macro `WRIST_MK2_RIGHT` to create binary for the right wrist
- remember to update the application version so that you can discriminate amongst which versions are used.

pattacini, 2023-03-02:
> Hi @pattacini, if the flashing procedure is the same (via Firmware updater) we can do it as soon as you provide us with the firmware. (It would be really helpful if you give us also the IP of the boards we have to flash so we are sure we do not make any mistakes)

@ale-git and @sgiraz need first to yield the AMC binaries (following @marcoaccame's [recommendations](https://github.com/robotology/icub-tech-support/issues/1518#issuecomment-1451588642)). Then, @GiulioRomualdi, you can get aligned with them.

pattacini, 2023-03-02:
Quick update on this.
https://github.com/robotology/icub-firmware/pull/351 doesn't seem enough. @ale-git and @sgiraz are debugging and testing the code on our setup. Once done, they'll provide the binaries for flashing the patch on the robot. @sgiraz may assist during the flash, if needed.

pattacini, 2023-03-02:
New PR:
- https://github.com/robotology/icub-firmware/pull/352.

pattacini, 2023-03-02:
I think we can close this task.
Feel free to open it up again if required.

closed 2023-03-02 by pattacini