robotology/mesh-support #1519: Burning smell around right shoulder

lrapetti, 2023-03-02:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While using the robot we noticed decreased performances, and we smelled burning. We immediately stopped the robot and turned off the battery

### Detailed context

We were performing teleoperated walking

sgiraz, 2023-03-02:
☝🏻 Hi @lrapetti, may you paste the log of `yarprobotinterface` in the attachment here, please?

---

@AntonioConsilvio, on the right forearm we have to verify if there are faults on:
- motors 
- `amcbldc` boards
- `aea` encoders

Check also the elbow motors, please (probably we may use the jogger)

cc @MrAndrea @maggia80 @pattacini

sgiraz, 2023-03-06:
@AntonioConsilvio checked the values of the resistors on the motor phases and it seems that a motor is damaged. The same one as the previous times.

sgiraz, 2023-03-06:
Closing in favour of #1520

closed 2023-03-06 by sgiraz