robotology/mesh-support #1520: Right wrist not working

lrapetti, 2023-03-02:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While using the robot we noticed the right wrist was not reading the encoder.

### Detailed context

We noticed it while using the robot in position control.

sgiraz, 2023-03-02:
Hi @lrapetti, may you paste the log of `yarprobotinterface` in the attachment here, please?

@AntonioConsilvio, on the right forearm we have to verify if there are faults on:
- motors 
- `amcbldc` boards
- `aea` encoders

Check also the elbow motors, please (probably we may use the jogger)

cc @MrAndrea @maggia80

DanielePucci, 2023-03-03:
After https://github.com/robotology/icub-tech-support/issues/1523 and https://github.com/robotology/icub-tech-support/issues/1522 we tried to restart the robot. 

We noticed that the right wrist is still not working: all joint encoders measure zero, at the motor side instead, we read them

sgiraz, 2023-03-06:
> While using the robot we noticed the right wrist moved to the limit and it was not reading the encoder.

There are two possible reasons why this occurs: either the FW running on the `AMC` board does not include the coupled joint logic (wrong FW flashed) or the calibration values inside the calibrators are incorrect.
At the moment the correct FW to have on the AMC board is v101.0 on the right arm and v102.0 on the left arm.

---
**Notes:**
After https://github.com/icub-tech-iit/proto/issues/2101 and https://github.com/icub-tech-iit/proto/issues/2131 the same motor has been replaced for the 3rd time. The new one worked for less than 24 hours.
We need to further investigate this issue!

cc @maggia80 @fbiggi @Fabrizio69 @AntonioConsilvio @Nicogene @pattacini

DanielePucci, 2023-03-11:
@sgiraz we were wondering if there were any update about this issue that is currently blocking some of the on-going tests for payload carrying https://github.com/ami-iit/element_walking-with-payload/issues/47#issuecomment-1463889032

sgiraz, 2023-03-11:
Hi @DanielePucci, next week we will unmount the right forearm for motor replacement. I cannot give you an exact date when the forearm will be available again, but I'll keep you updated.

cc @AntonioConsilvio @Fabrizio69

sgiraz, 2023-03-30:
Hi @DanielePucci,

The problem we found was caused by the magnetic interference between the hall sensors of the `AMCBLDC` (the one that controls the M0 motor) and the motor placed under the board. 
We actually flipped that board and replaced the faulted motor. The right wrist should now work correctly!

**Note:** We also applied the same workaround on the left wrist.

We are waiting for your feedback.

cc @AntonioConsilvio

DanielePucci, 2023-03-30:
Thanks @sgiraz , we'll get back to you @lrapetti

lrapetti, 2023-03-31:
Yesterday we did a few tests on the robot and we did not spot any issue with the wrist (we moved only the wrist yaw). 

For me we can close this issue, and eventually re-open a new one if we experience again issues

sgiraz, 2023-03-31:
Ok @lrapetti, thanks!

closed 2023-03-31 by sgiraz