robotology/mesh-support #1523: Right index open-close hits the mechanics when abduction is zero

DanielePucci, 2023-03-03:
### Device name 🤖

ergoCubSN000

### Request/Failure description

Before running the teleoperation, we were checking the hand motions. Because of https://github.com/robotology/icub-tech-support/issues/1522, the right-hand abduction joint is controlled and brought to zero. 

Then, we were trying to make the index open-close move, but it did not. While investigating, we notice that it hits the mechanics 

https://user-images.githubusercontent.com/5045846/222786453-5a34f759-04ef-4d91-80a7-3d725efb316a.mp4

### Detailed context

None

sgiraz, 2023-03-06:
> By idling the abduction and putting it to the center, then the finger no longer hits the mechanics. But I guess this is an issue anyways

Hi @DanielePucci, yep this is a known issue: https://github.com/icub-tech-iit/proto/issues/2071 

Anyway, @fbiggi is working on it!

cc @Fabrizio69 @fiorisi

DanielePucci, 2023-03-09:
Thanks @sgiraz 
> Hi @DanielePucci, yep this is a known issue

It may be worth to add the link to the issue you are referring to then

DanielePucci, 2023-03-30:
Cool, we may link the calibration issue then @sgiraz

closed 2023-03-31 by sgiraz

pattacini, 2023-03-31:
Our issue is internal, thus @sgiraz made the link internal -> this issue.
Whoever has access is able to see the reference.