robotology/mesh-support #1525: iCubGenova11 - Right hand does not complete calibration

xEnVrE, 2023-03-06:
### Device name 🤖

iCubGenova11

### Request/Failure description

The right hand does not complete the calibration. It remains as in the picture:

[image]

### Detailed context

It seems to me that the thumb opposition is not moving.

### Additional context

Please find here attached the log from the `yarprobotinterface`.

[log_icub-head_yarprobotinterface_2401.txt](https://github.com/robotology/icub-tech-support/files/10900784/log_icub-head_yarprobotinterface_2401.txt)

sgiraz, 2023-03-06:
There are many errors inside the log attached.
Some of the first errors are the followings:

```console
 32688,226071 <INFO> EthResource::askBoardVersion() found BOARD head-eb20-j0_1 @ IP 10.0.1.20 of type mc4plus with FW = ver 3.56 built on 2022 Dec 07 17:34
 32688,256822 <DEBUG>  from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 1894s 857m 412u: (code 0x05000015, par16 0x2000 par64 0x0000000000000000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
```

```console
 32689,556745 <ERROR>  from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 1896s 161m 190u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
```

> It seems to me that the thumb opposition is not moving.

@xEnVrE, have you tried to run the `yarprobotinterface` excluding the right arm?

xEnVrE, 2023-03-06:
Hi @sgiraz,

I can try and attach the log if useful!

xEnVrE, 2023-03-07:
I started the left arm alone and together with @sgiraz we spotted the following error on the left arm:

[image]

xEnVrE, 2023-03-07:
We checked together with @sgiraz and @AntonioConsilvio an we realised that the issue is coming from the right thumb. Specifically, the thumb proximal motor spins indefinitely without moving the joint.

AntonioConsilvio, 2023-03-08:
Hi @xEnVrE the proximal thumb motor was damaged and it was replaced by @GiacomoVassallo13. I noticed that the PID values of the right thumb were different from those of the left thumb, there is already a [PR](https://github.com/robotology/robots-configuration/pull/452) to change them.

The finger has been tested, however give us feedback on the operation of the finger when you will test it.

cc @sgiraz @fedeceola

sgiraz, 2023-03-08:
Well done @AntonioConsilvio, just to clarify, the test has been done using the same PID values of the left thumb, correct?

xEnVrE, 2023-03-09:
Today I tested the robot and everything seemed ok. I just noticed that the right thumb is missing the plastic ring that blocks the encoders cable. Can we add it again? @sgiraz 

thank you

AntonioConsilvio, 2023-03-10:
Hi @xEnVrE, the plastic ring has been put back on the thumb. The intervention is complete.

cc @sgiraz

xEnVrE, 2023-03-10:
Thanks @sgiraz @AntonioConsilvio

xEnVrE, 2023-03-10:
Should I close this?

sgiraz, 2023-03-10:
Thanks for your feedback @xEnVrE.
Closing!

closed 2023-03-10 by sgiraz