robotology/mesh-support #1526: Battery stops charging

lrapetti, 2023-03-07:
### Device name 🤖

ergoCubSN000

### Request/Failure description

 When charging the battery, we noticed that after a while it stops absorbing energy.

### Detailed context

Turning on and off the battery, it restarts to charge it.

maggia80, 2023-03-07:
Cc: @MrAndrea

MrAndrea, 2023-03-08:
Ciao @lrapetti, do you remember the power supply settings (voltage, current) when this problem occurred?

lrapetti, 2023-03-09:
> Ciao @lrapetti, do you remember the power supply settings (voltage, current) when this problem occurred?

It is the same power supply we use for iCub3, and I think nobody touched the settings.

cc @S-Dafarra @GiulioRomualdi

MrAndrea, 2023-04-11:
Strangely the power supply was wrong, in fact it was the _EA-PSI 9040-40 T_ which has 40V / 40A / 1kW maximum output:

| [image] | [image] |
|:-:|:-:|

It has been replaced with the _EA-PS 9080-60 T_ which has 80V / 60A / 1.5kW maximum output:

| [image] | [image] |
|:-:|:-:|

## SETTINGS

| Output voltage | 44V |
|:-|:-|
| Maximum output current | 25A |
| Maximum power | 1200W|
| Undervoltage limit | 42V |
| Overvoltage limit | 45V |
| Overcurrent limit | 30A |

MrAndrea, 2023-04-11:
The battery charging has the expected behavior and the problem happened because the previous power supply couldn't give more than 40V, and the charger was set to 42V. This thing created a conflict between the two power supplies and the battery.

This issue can be closed.

closed 2023-04-11 by MrAndrea

lrapetti, 2023-04-11:
Hi @MrAndrea, thanks for fixing the issue.

Since the same charger is/was used also for the iCub3 battery, would this cause any issue in case we want to use it again for iCub3?

maggia80, 2023-04-11:
> Hi @MrAndrea, thanks for fixing the issue.
> 
> Since the same charger is/was used also for the iCub3 battery, would this cause any issue in case we want to use it again for iCub3?

iCub3 power source is version 9080-60 T and it was set to 44V.

MrAndrea, 2023-04-11:
> Since the same charger is/was used also for the iCub3 battery, would this cause any issue in case we want to use it again for iCub3?

Yes it can happen the same problem because iCub3 has the same electrical characteristics as ergoCub, don't use that model.

S-Dafarra, 2023-04-11:
> Since the same charger is/was used also for the iCub3 battery, would this cause any issue in case we want to use it again for iCub3?

Also, note that, if I understood correctly, the problem was not in the battery charger, but in the power supply.

MrAndrea, 2023-05-15:
New consideration on the battery charging problem at [this comment](https://github.com/icub-tech-iit/electronics-boards/issues/325#issuecomment-1547282863).

S-Dafarra, 2023-05-15:
> New consideration on the battery charging problem at [this comment](https://github.com/icub-tech-iit/electronics-boards/issues/325#issuecomment-1547282863).

I don't think that is publicly accessible

MrAndrea, 2023-05-15:
> I don't think that is publicly accessible

I drop here the comment:

---

As speculated by the BMS designer, it could be a temperature problem.

>Buongiorno Andrea,
il BMS solitamente esclude il caricabatterie se c'è una qualche situazione di fault. La cosa più classica da come mi hai descritto il comportamento è un overtemperature in carica. 
Possibile che la temperatura delle celle salga sopra i 40°C ? Tipicamente accade se tentate di ricaricare la batteria subito dopo la scarica. In questo caso il BMS va ad interrompere la carica ma la scarica resta possibile in quanto la batteria può operare in scarica fino a 60°C.
Se fosse questa la causa l'unica soluzione è prescrivere di lasciar riposare la batteria dopo il ciclo di scarica un tempo opportuno affinchè si raffreddi a sufficienza per non andare in blocco nella successiva carica.
Daniele

S-Dafarra, 2023-05-15:
> As speculated by the BMS designer, it could be a temperature problem.

Do you think it could be possible to install a fan to reduce the temperature of the battery? My concern is that often it is not possible to "wait". We already have been in situations where the robot has to be on, waiting for a demo, while standing. In these cases, it is crucial to be able to recharge the battery until the very last moment.

MrAndrea, 2023-05-15:
I don't know if there is enough space to put a fan close to the battery.
I'm asking the BMS designer if there is a possibility to do something like extend the temperature range or other.

MrAndrea, 2023-05-15:
Here is the reply of the designer after asking for some opinions.

>Purtroppo non possiamo alzare i limiti perchè le soglie in temperatura sono settate nei limiti di sicurezza prescritti dal costruttore per il tipo di cella impiegato nel vostro pacco batterie.
Si potrebbero valutare celle che consentono di arrivare in carica un pò più in alto in temperatura se la ventola non dovesse risolvere la questione, ma nel caso per questa esigenza vi dovete confrontare con ELTEC per la realizzazione di una batteria differente.

MrAndrea, 2023-05-15:
I received another clarification email:

>Aggiungo solo che le batterie al litio non nascono per rimanere in tampone quindi non dovrebbero essere ricaricate mentre vengono scaricate, anche per questioni legate alla safety e per l'invecchiamento delle batterie. Il BMS interviene ad impedire la ricarica in situazioni di pericolo ma concettualmente è  sbagliato utilizzare la batteria in questo modo d'uso.
Quindi sarebbe meglio evitare situazioni come quella descritta, se possibille.

S-Dafarra, 2023-05-15:
> I received another clarification email:

Thanks for the clarification. In principle, when we charge the battery, we also connect the power supply, so in principle that situation should be less likely to happen (although it did happen for sure in the past). The situation I was thinking about in the previous comment was the following:
1. We did some pre-demo tests with the robot moving (and walking) using only the battery. Hence, the battery drains quite quickly and heats up. These tests usually need to run pretty close to the demo start because of time constraints.
2. We stop the tests, waiting for the demo to start (imagine the arrival of guests, or the robot waiting on a stage). We need to connect the robot to a power supply (otherwise it won't last for the demo) and also recharge the battery (because of the previous tests). We cannot shut the robot down because it is too risky (some board may fail to start, calibration issues, we would need to reboot too many components)
3. The demo starts, usually with just a couple of minutes of forewarning

Clearly, the "wait and let it cool down" strategy cannot work, due to possible time constraints. Then, which other strategies are we left with?

MrAndrea, 2023-06-01:
> Clearly, the "wait and let it cool down" strategy cannot work, due to possible time constraints. Then, which other strategies are we left with?

Fans close to the BMS and holes in the cover have been added, see:
- https://github.com/icub-tech-iit/proto/issues/2260
- https://github.com/icub-tech-iit/robot-snapshots/issues/34

S-Dafarra, 2023-06-02:
> > Clearly, the "wait and let it cool down" strategy cannot work, due to possible time constraints. Then, which other strategies are we left with?
> 
> Fans close to the BMS and holes in the cover have been added, see:
> 
> * https://github.com/icub-tech-iit/proto/issues/2260
> * [[ergoCub_SN_000] [Implementation for better cooling on the upper body] icub-tech-iit/robot-snapshots#34](https://github.com/icub-tech-iit/robot-snapshots/issues/34)

Great, thanks a lot. We did not have any issue during ICRA

MrAndrea, 2023-06-05:
I think that this issue can be closed.