robotology/mesh-support #1528: Motor temperature suspiciously high around some joints

DanielePucci, 2023-03-12:
### Device name 🤖

ergoCubSN000

### Request/Failure description

The ergoCub robot, since it's delivery, has been experiencing high temperatures of some motors. This happens, for instance, for the motors of the hip rolls. 

This is a sort of new phenomenon with respect to iCub3, which did not show the temperature issue. One of the assumptions was that the robot arms weigh more than those of iCub3 - same motors at the shoulders, but not experiencing the temperature issue - and this  would explain higher temperatures.

Clearly, this investigation should be done more carefully by looking at, for instance, motor currents. 

At the qualitative level, the iRonCub team @HosameldinMohamed @vpunithreddy @gabrielenava is currently performing the in-situ calibration of the iCub3 FT sensors at the arm https://github.com/ami-iit/element_ft-nonlinear-modeling/issues/66 and for this reason it installed weights on its arms. 

And the temperature of the shoulder rolls on iCub3 stays relatively low even after some time with a relatively high weight. 
[image]

CC 
@lrapetti 

### Detailed context

-

maggia80, 2023-03-13:
thanks @DanielePucci for reporting this. Yes, I agree with you we need to further investigate the behavior on ergoCub, starting from shoulder roll.

DanielePucci, 2023-03-20:
Some other interesting experiments are collected at https://github.com/ami-iit/element_ft-nonlinear-modeling/issues/71

DanielePucci, 2023-03-29:
While @GiulioRomualdi was performing the IMU location https://github.com/icub-tech-iit/tickets/issues/2684#issuecomment-1486890876, I passed by and I checked the motor temperature not only of the shoulders, but **but also of the torso**.

The temperature of the torso joint, to which we attach the temperature, is very high: I could not touch the plate behind the IMU - see below - for more than 2 seconds, and the robot walked only once ⚠️ So, I do not know if temperature effects may alter the IMU reading in the long run 

[image]

CC
@sgiraz @pattacini @maggia80

MrAndrea, 2023-04-24:
A motor had burned out. See: 
- https://github.com/icub-tech-iit/proto/issues/2204#issuecomment-1504919244
- https://github.com/icub-tech-iit/proto/issues/2202#issuecomment-1503722591

sgiraz, 2023-05-07:
Hi @DanielePucci,

Are you still noticing high temperatures around the indicated joints?? If not, I guess we can close this issue.

DanielePucci, 2023-05-09:
Yes, during yesterday demo https://github.com/ami-iit/lab-events-demos/issues/248 the shoulder motors were really at high temperature

sgiraz, 2023-05-17:
We have reason to believe that the perceived heat on the shoulder joints may be generated by poor dissipation of the main torso board. We are implementing a solution to achieve better cooling. 

[image]

**N.B.** This is only a temporary workaround for `ergoCubSN000`

cc @maggia80

sgiraz, 2023-06-05:
It seems that the new cooling fan system gets down the temperature of the joints mentioned above. Afaik on the next ergoCub the cooling system will be improved furtherly.
Closing, feel free to re-open if necessary.

closed 2023-06-05 by sgiraz