robotology/mesh-support #1529: ergoCubSN000 - Cannot use fingers of the left hand

xEnVrE, 2023-03-15:
### Device name 🤖

ergoCubSN000

### Request/Failure description

We were tried to use the fingers of the left hand of the robot. We are using the latest configuration of the robot from robots-configuration/master.

After running the `yarprobotinterface` the fingers are either in hardware fault as the fault is not detected or not configured:

[image]

### Detailed context

I am attaching the log file of the `yarprobotinterface`.

[log_ergocub-torso_yarprobotinterface_4874.txt](https://github.com/robotology/icub-tech-support/files/10982721/log_ergocub-torso_yarprobotinterface_4874.txt)

xEnVrE, 2023-03-15:
As a side note, in order to run the robot we had to - with the help of @AntonioConsilvio - change the following line in the robots configuration files associated to the left wrist:

```console
diff --git a/ergoCubSN000/hardware/motorControl/left_arm-eb31-j4_6-mc_service.xml b/ergoCubSN000/hardware/motorControl/left_arm-eb31-j4_6-mc_service.xml
index cae67802b..1665e2637 100644
--- a/ergoCubSN000/hardware/motorControl/left_arm-eb31-j4_6-mc_service.xml
+++ b/ergoCubSN000/hardware/motorControl/left_arm-eb31-j4_6-mc_service.xml
@@ -22,7 +22,7 @@
                 <group name="FIRMWARE">
                     <param name="major">            1           </param>
                     <param name="minor">            0           </param>
-                    <param name="build">            10           </param>
+                    <param name="build">            9           </param>
                 </group>
             </group>
```

Not sure if this has to be tracked somewhere else.

cc @sgiraz @pattacini

pattacini, 2023-03-16:
>As a side note, in order to run the robot we had to - with the help of @AntonioConsilvio - change the following line in the robots configuration files associated to the left wrist:

We have delivered a fresh release for the wrist very recently.
Perhaps, @mfussi66 can have a look at it.

AntonioConsilvio, 2023-03-16:
Hi @xEnVrE, the problem was that the new [distro](https://github.com/orgs/robotology/discussions/613 ) contains an update for a new calibration for eCub fingers (calibration 6). 

This calibration needs changes within RobotsConfiguration and a new Firmware for the MC4-PLUS, contained in these two PRs ([ PR Robots config](https://github.com/robotology/robots-configuration/pull/454), [PR icub firmware build](https://github.com/robotology/icub-firmware-build/pull/71)).

Only the RobotsConfiguration was updated in the eCub torso. Therefore, the robot was not aligned.

However, at this time the Git local history was rewritten by removing the commit that enabled calibration 6 because the hardware is not ready to perform this new calibration.

So there are divergence of one commit between the repository in ergocub pc and github.

Now the fingers are working as before. Regarding the index abduction needs hardware improvement to be used so it fall in Hardware fault, and this HW fault sends index Finger OpenClose to idle.

The thumb motor encoder cannot be read, so it can send the thumb to idle.

It is therefore necessary run the fingers manually from yarpmotorgui to use them.

cc @maggia80 @pattacini @mfussi66 @lrapetti @GiulioRomualdi

xEnVrE, 2023-03-16:
Thanks @AntonioConsilvio @mfussi66 

I think this can be closed.

closed 2023-03-16 by pattacini