robotology/mesh-support #1530: Right forearm tendon broken

ddetommaso, 2023-03-16:
### Device name 🤖

iCubErzelli02

### Request/Failure description

A single tendon of the right forearm broke during manipulation operations.

### Detailed context

no more details

### Additional context

[image]
[image]

### How does it affect you?

We were in the final phase of testing an experiment that should run in the next couple of days. Each day without the robot working may have a severe impact on delaying the data collection already scheduled.

GiacomoVassallo13, 2023-03-16:
Good morning, with my colleagues and my manager it was decided to intervene tomorrow morning to restore the elbow.
I will arrive at about 0900. I kindly ask you to let me know the location of the robot, so that I don't have to look for it in case there is no one to show it.

cc: @sgiraz @AntonioConsilvio @Fabrizio69

ddetommaso, 2023-03-16:
Hi @GiacomoVassallo13 

you can contact me, I'll be at IIT tomorrow. 
Anyway, the robot is located on the 7th floor in the room 07-ST19.0 (out from the elevators take the corridor on the right).
The laptop is already connected to the robot and working with YARP on a docker container (in the desktop there is a folder named 'icub-tech' where you'll find instructions on how working with the robot).

Thank you

AntonioConsilvio, 2023-03-17:
Hi @ddetommaso, the intervention carried out by @GiacomoVassallo13 should have been successful, however give us feedback on the operation of the robot when you can.

cc @sgiraz @Fabrizio69

ddetommaso, 2023-03-17:
Hi guys,

we can close the issue! Many thanks for the very quick support.
Well done!

cc @AgnieszkaWykowska

closed 2023-03-20 by sgiraz