robotology/mesh-support #1531: Torso tendons are very loose

gabrielenava, 2023-03-21:
### Device name 🤖

iCubGenova04

### Request/Failure description

the torso tendons are very loose. the robot in position control oscillates a lot as soon as it is touched. At a first look, it does not seem there is something broken.

### Detailed context

https://user-images.githubusercontent.com/12396934/226630053-4c011519-705e-4a2f-8b1d-908937ff0d98.mp4

GiacomoVassallo13, 2023-03-24:
Good morning, the tendon in the torso has been tensioned. To do this, the backpack had to be disassembled. The robot has been returned.

sgiraz, 2023-03-24:
Well done @GiacomoVassallo13! 🚀
I think we can close this issue.

closed 2023-03-24 by sgiraz