robotology/mesh-support #1532: software/firmware upgrade of talking Head iCub (with Atom PC104)

EliseiFrederic, 2023-03-21:
### Device name 🤖

iCubGrenoble01

### Request/Failure description

We plan (for network security reasons) to do a software/firmware upgrade of iCubGrenoble01 (Atom PC104 iCub + talking head)
Any specificity/warning/incompatibility that we should be aware of before starting ?
Maybe the Live Debian 10 distro with Atom ? the slow network that was experiment previously and forced a modified Debian 9 distro ?  firmware compatibility or new formats for the robot files ?

### Detailed context

We currently have 
7.9 iCub LIVE image from Debian 9/stretch on PC104
YARP 3.2 on Debian 8 on icubsrv and cluster
probably an upgrade from december 2017

### How does it affect you?

Just want to be sure we can start the ugrade !
Robot bazaar describes 2.1 Talking Head as "obsolete"
Still works almost fine from our point of view :-)

AntonioConsilvio, 2023-03-31:
Hi @EliseiFrederic, We received your issue. We will provide the best support as soon as possible, stay tuned!

cc @sgiraz

EliseiFrederic, 2023-04-20:
As a new baby, ergoCub requests much attention it seems...

sgiraz, 2023-04-24:
Hi @EliseiFrederic,

> As a new baby, ergoCub requests much attention it seems...

Yep, as it has been churned out we had to solve many high-priority issues due to the imminent demos and deadlines.
Anyway, we recently opened a discussion in https://github.com/orgs/robotology/discussions/614 that should be helpful for your request as well. Please, try to follow the same suggestions and let us know. 

cc @AntonioConsilvio @Fabrizio69

EliseiFrederic, 2023-04-25:
Thanks for pointing to the discussion. We always use a second icubsrv and second USB key when upgrading, as it may takes ages before it's back to "all working" with the new enhancements :-)
But with the talking head version, keeping the head opened is not very convenient. And every time you close it again, you might have to slightly change limit values of the frontface moving elements...

EliseiFrederic, 2024-12-06:
Upgrade being done at iCub facility...

closed 2024-12-06 by EliseiFrederic