robotology/mesh-support #1533: iCubAberystwyth01 - Software upgrade

PHShaw, 2023-03-28:
### Device name 🤖

iCubAberystwyth01

### Request/Failure description

After having been out of operation for several years, we're trying to resume using the iCub.  The usb in the head seems to have failed so we're not able to boot the previous build that we had operational.

### Detailed context

We have a new computer and a new usb for the head, so will be a clean build on both systems.  What additional customization do we need to do to make the software side operational?

### How does it affect you?

We are in the process of applying for funding that will rely on the iCub hardware, so want to make sure it is still operational before proceeding.

AntonioConsilvio, 2023-03-31:
Hi @PHShaw, We received your issue. We will provide support as soon as possible, stay tuned!

cc @sgiraz

sgiraz, 2023-05-12:
Hi @PHShaw,

We recently opened a discussion in https://github.com/orgs/robotology/discussions/614 that should be helpful for your request as well. Please, try to follow the same suggestions and let us know.

closed 2023-09-27 by github-actions[bot]