robotology/mesh-support #1534: Left wrist is not working as expeced

sgiraz, 2023-03-29:
### Device name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

After mounting the right arm and rotating the amcbldc `e31b0`, we tried to test the two forearms independently. The left wrist is not starting correctly.

### Detailed context

Here is the log of the `yarprobotinterface` during the startup:

[log_290323_1429.txt](https://github.com/robotology/icub-tech-support/files/11101936/log_290323_1429.txt)

### Additional context

Remount the AMCBCLDC to its original position and see what happens. If it doesn't work, we will have to re-check the wires and connectors otherwise it could be a short circuit somewhere when upside down.

### How does it affect you?

Needed for `ICRA!`

sgiraz, 2023-03-29:
cc @Gandoo @Fabrizio69

maggia80, 2023-03-30:
Did you have the kapton tape on the motor under the AMC-BLDC in the left wrist?

sgiraz, 2023-03-30:
@AntonioConsilvio and @Gandoo found the hall sensors wires broken again. This time they used some ready-to-use crimped wires and soldered them to the one coming from the motor.
The problem seems to be fixed for now.

closed 2023-03-31 by sgiraz