robotology/mesh-support #1535: R1 Power Cable not working

randaz81, 2023-03-30:
### Device name 🤖

R1 SN001 Erzelli

### Request/Failure description

The robot is not properly charging its battery unless the male connector is pressed tightly into the robot plug.
So either the female connector on the robot, or the male plug on the cable from the power supply is damaged.

### Detailed context

---

maggia80, 2023-03-30:
@randaz81 is the battery still capable of powering the robot? Otherwise could be the 20A fuse in series with the battery broken.

randaz81, 2023-03-30:
yes. It seems that just the connector (male/female i do not know) is loose. Everything works if I keep the cable pushed in.

Fabrizio69, 2023-04-11:
Hi @randaz81 ,
tomorrow morning @AntonioConsilvio will be in Erzelli to repair the charging cable. Thanks

cc: @sgiraz

AntonioConsilvio, 2023-04-12:
Hello @randaz81, I solved the problem today. The problem was caused by the female connector. Now everything seems to be working properly.

https://user-images.githubusercontent.com/114915464/231489318-8ff769a5-8bfb-4271-8fca-86ef994b1c99.mp4

cc @sgiraz @Fabrizio69 @maggia80

randaz81, 2023-04-12:
Thank you! We are going to check it asap!

AntonioConsilvio, 2023-04-12:
Perfect. I await your feedback! @randaz81

Fabrizio69, 2023-04-14:
Hi @randaz81, this was an urgent support request, do you have any feedback on this?

cc: @AntonioConsilvio

randaz81, 2023-05-02:
The problem has been successfully fixed, thank you!

closed 2023-05-02 by randaz81