robotology/mesh-support #1537: iCub3 left shoulder roll jerky and noisy in idle

DanielePucci, 2023-04-05:
### Device name 🤖

iCubV3

### Request/Failure description

Today, I started touching the robot at random, and I noticed with @gabrielenava that the left shoulder roll is a little noisy and jerky when in idle, see and hear the video below

### Detailed context

See above

### Additional context

https://user-images.githubusercontent.com/5045846/230048729-16f0c103-896c-4691-8d75-63d09e400f56.mp4

gabrielenava, 2023-04-05:
there is also the left leg roll which is moving in a very scattered way during calibration, and it makes also a strange noise:

https://user-images.githubusercontent.com/12396934/230153913-b1d22618-7fc0-49a1-848d-c18f4a1bae9a.mp4

would it be possible to check also this joint?

DanielePucci, 2023-04-06:
Ok @gabrielenava, I updated the issue title

gabrielenava, 2023-05-04:
we tested the robot today, it seemed fine to me for what concerns this issue!

I have opened dedicated issues for other problems that emerged during the testing #1551 #1552

sgiraz, 2023-05-05:
Ok, I'm closing this one then.