robotology/mesh-support #1542: left shoulder cable 8 broke during reaching action

reafrancesco, 2023-04-20:
### Device name 🤖

iCubGenova07

### Request/Failure description

left shoulder cable 8 broke during reaching action.

### Detailed context

left shoulder cable 8 broke during reaching action.
The shoulder was fixed and only n.1 of cable 8 is left in the spare set.

Considering the frequency of such issue it is worth considering ad-hoc inspection of the left shoulder.

sgiraz, 2023-04-24:
Hi @reafrancesco,

This should be the tendon you mean, right?

[[image]][1]

We'll discuss your request internally and let you know asap.

cc @GiacomoVassallo13 @fbiggi @Fabrizio69 

[1]: https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/#2-left-shoulder

reafrancesco, 2023-04-26:
Yes this is the correct one and obviously we need some spare ones since we have repaired it frequently in the last months.
Thanks,
Francesco

sgiraz, 2023-04-27:
Ok, we are going to send you spare cables type-7 and type-8 as requested.
Therefore, we plan to do a check to understand the possible reason for this frequent failure.

cc @Fabrizio69

sgiraz, 2023-05-05:
Hi @reafrancesco, 

Today we are going to ship the tendons. You should receive them next week.

cc @AntonioConsilvio

sgiraz, 2023-05-25:
Given the absence of feedback, I assume you have received the tendons.
Closing.

closed 2023-05-25 by sgiraz