robotology/mesh-support #1544: [ergoCubSN000] Several robot startup failing with different joints not calibrating

lrapetti, 2023-04-20:
### Device name 🤖

ergoCubSN000

### Request/Failure description

After mounting the torso pitch pin (see https://github.com/robotology/icub-tech-support/issues/1538#issuecomment-1506989419) and removing the torso pitch from the calibrator, we are not able to start the robot because joints are not failing (every time the failure seems to be in different joints).

### Detailed context

Here are a couple of videos of the calibration with the associated log:

# Video 1

https://user-images.githubusercontent.com/35487806/233426423-a5dcbcca-a0cb-4550-a2ab-3a198aab5f8b.MOV

[video_1.txt](https://github.com/robotology/icub-tech-support/files/11288323/video_1.txt)

# Video 2

https://user-images.githubusercontent.com/35487806/233426518-852f46cf-95f6-40eb-a7ad-788687c84283.MOV

[video_2.txt](https://github.com/robotology/icub-tech-support/files/11288329/video_2.txt)

# Other logs
Here is a txt with also a few other startup attempts:

[error.txt.zip](https://github.com/robotology/icub-tech-support/files/11288335/error.txt.zip)

sgiraz, 2023-04-20:
Hi @lrapetti, can you try to restart both the motors and the head, then try to run the `yarprobotinterface` again, please?

lrapetti, 2023-04-20:
> Hi @lrapetti, can you try to restart both the motors and the head, then try to run the `yarprobotinterface` again, please?

We tried earlier, but we gave another try  and the right hip is not configured (plus a few other joints in HW)

cc @DanielePucci

lrapetti, 2023-04-20:
> > Hi @lrapetti, can you try to restart both the motors and the head, then try to run the `yarprobotinterface` again, please?
> 
> We tried earlier, but we gave another try and the right hip is not configured (plus a few other joints in HW)

I don't know if this error is related to the issue -> https://github.com/robotology/icub-tech-support/issues/1543

lrapetti, 2023-04-21:
The robot was tested by @GiulioRomualdi, I think he did not spot any issue but he can confirm

sgiraz, 2023-04-23:
Good job guys! 
I guess this issue can be closed.  Feel free to reopen if necessary.

closed 2023-04-23 by sgiraz