robotology/mesh-support #1546: ergoCubSN000 - Neck pitch tracking not ideal

xEnVrE, 2023-04-27:
### Device name 🤖

ergoCubSN000

### Request/Failure description

We would like to know if it is possible to refine the tuning (in terms of PID and/or mechanics) of the neck pitch joint.

### Detailed context

While testing a gaze controller, we observed that the joint does not properly track the desired reference, sent using `position direct`.

[image]

As it can be seen, the actual joint motion is lagging behind the desired reference.

We also attach a video of the neck moving (please take into account that also the neck roll and yaw were moving in this experiment):

https://user-images.githubusercontent.com/6014499/234949275-b902886a-046c-4e8b-8b23-c5cd8c43581a.mp4

xEnVrE, 2023-04-27:
cc @andrearosasco @sgiraz

pattacini, 2023-04-27:
Hi @xEnVrE 

This is somewhat expected as we significantly detuned the PID gains in order to prevent the neck from going into instability during the highly dynamic tasks required by the AMI group (i.e., walking, lifting).

Investigations are still ongoing to come up with a good set of gains for multiple scenarios.

If you aim to carry out your tests with the robot standing still and upright, you may **temporarily** increase the gains as per below:

[image]

⚠️ Please, remember to bring them back to their original **detuned** values once done with your experiments.

| Examples of tracking |
| :---: |
| [image] |
| with detuned gains |
| [image] |
| with the new gains |

xEnVrE, 2023-04-27:
Hi @pattacini,

thanks for the precious information. We will make a test with the suggested PID values, _and move them back once done_, so that we can continue with our activities. 

During the demo at ICRA 2023 I guess we could change the PID values depending on who is operating the robot.

xEnVrE, 2023-04-28:
We tested with new PID values and now the tracking of the pitch (green signals) seems better - for sure less lag than before.

[image]

We are reporting also the other neck joints and the realsense tilt since we collected them.

The overall behavior while tracking a circular path in 3D space is as follows:

https://user-images.githubusercontent.com/6014499/235182387-dd9d0946-6460-4795-acbd-ddc47003d6d5.mp4

Although a bit woobly, we are satisfied with that for now. Hence I think we could close the issue.

closed 2023-04-30 by sgiraz

valegagge, 2023-05-02:
Hi @xEnVrE ,

> During the demo at ICRA 2023 I guess we could change the PID values depending on who is operating the robot.

The motion control configuration file lets the user define more pid groups and activate one per time.
currently, it isn't possible to swap the PID values on the fly unless setting the new values by a user application, but you can store different PID values on the same file avoiding swapping between robot-configuration branches. 

You can find more information on the [icub-tech-documentation](https://icub-tech-iit.github.io/documentation/icub_robot_configuration/devices_configuration/devices_configuration/#motor-control-on-icub).

I know this isn't the optimal solution... but it can reduce your efforts :) .....

xEnVrE, 2023-05-02:
Thanks for the suggestion @valegagge.

I think we will change/restore them using the `yarpmotorgui` or from our application using the proper API.

> The motion control configuration file lets the user define more pid groups and activate one per time.

Would this require restarting the `yarprobotinterface` to select the other group?

Thank you

valegagge, 2023-05-02:
> Would this require restarting the `yarprobotinterface` to select the other group?

sadly yes