robotology/mesh-support #1547: [ergoCubSN000] Unexpected behaviour when switching from compliant to stiff  (and vice versa) with joint in position direct

GiulioRomualdi, 2023-05-02:
### Device name 🤖

ergoCubSN000

### Request/Failure description

⚠️ Hi! I'm opening the issue here as an entry point, if you think there is a better place to open this issue I can close this one and reopen it in another place.

While testing the compliant mode on the ergoCub we noticed a strange behavior. Indeed when switching from compliant to stiff and or vice versa when the motor is controlled in position direct makes the joint stiff both in the sense of rigidity and viscosity, 

### Detailed context

The following video tries to explain better what we are experiencing 

1. the joint was set to position and compliant mode (the selection of the compliant mode was done when the joint was in position) In this case the joint is behaving as expected 🟢 
2. Then we switched from position to position direct and also here the joint was moving  as expected 🟢 
3. Finally we did the following
   1. switch from position direct to position
   2. switch from compliant to stiff
   3. switch from stiff to compliant
  In this configuration, the joint seems to be stiffer than before 🔴 

The experiment was done using the motorgui. 

https://user-images.githubusercontent.com/16744101/235675067-1ddadd53-3fb5-4950-b7c5-55910eef30e8.mp4

### Additional context

cc @S-Dafarra @mebbaid @lrapetti

sgiraz, 2023-05-17:
Hi @GiulioRomualdi, I think that these control modes are not so common like `position` (direct), `current` or `pwm.` Probably @ale-git is aware of what is happening here. I believe the best way to resolve this issue is to open a [discussion][1] in tickets.

cc @valegagge @Nicogene @pattacini 

[1]: https://github.com/icub-tech-iit/tickets/issues/new?assignees=maggia80&labels=request-discussion&projects=&template=request-discussion.md&title=

S-Dafarra, 2023-06-16:
We noticed that if we switch to compliant from position, the behavior is very stiff. If we switch to idle and back to position (remaining in compliant mode), then the behavior is good.

S-Dafarra, 2023-06-30:
We found another way to have it working
Joint in position stiff -> switch to compliant -> the compliance is terrible

Joint in position stiff -> switch to idle -> switch to compliant mode -> switch back to position -> the compliance is good.

I am pretty sure this is a firmware bug. @sgiraz @pattacini would you prefer to move this to a ticket?

pattacini, 2023-06-30:
Hi @S-Dafarra 

>I am pretty sure this is a firmware bug. @sgiraz @pattacini would you prefer to move this to a ticket?

I think @valegagge and @ale-git are following up on this.
Opening up a ticket on this would give just a little benefit. I have no strong opinion on that; probably, it's worth keeping all the discussion here.

S-Dafarra, 2023-09-19:
@pattacini @valegagge @ale-git any update on this?

pattacini, 2023-09-19:
Added https://github.com/robotology/icub-tech-support/labels/pinned.

pattacini, 2023-09-20:
Hi @S-Dafarra 

Team FIVE has an internal ticket tracking this activity, which has been delayed though, due to the need for testing ergoCubSN001.

valegagge, 2023-09-20:
Hi @S-Dafarra ,
we don't forget this issue, but we are involved in testing activity related to ergocub s/n 001.

S-Dafarra, 2024-05-17:
Hi everyone, from time to time, depending on demos, it is still useful to have joints in compliant mode, but this is blocking. Any update on this?

Nicogene, 2024-05-21:
Hi @S-Dafarra, we will start investigating the next week, we will update you

cc @pattacini @ale-git

Nicogene, 2024-05-29:
Hi @GiulioRomualdi, @ale-git should have found the source of problem:
>The motor input type (pwm, current or velocity) wasn't properly set during stiff <-> compliant transitions, and thus we had misbehaviour when the stiff control output type was different from torque control type as in the ErgoCub configuration.

>This should fix the problem (to be tested):
[ale-git/icub-firmware@compliant-bugfix](https://github.com/ale-git/icub-firmware/tree/compliant-bugfix?rgh-link-date=2024-05-29T07%3A26%3A34Z)

We will soon test the fix and let you know!

Nicogene, 2024-06-05:
Tomorrow we will test the @ale-git fix on `iCubGenova11` and then we will proceed with the PR in case of positive outcome

cc @GiulioRomualdi @pattacini

Nicogene, 2024-06-06:
@GiulioRomualdi the today's test gave a positive outcome, @ale-git will provide soon a PR with the fix!

GiulioRomualdi, 2024-06-06:
That's great! Thank you!

Nicogene, 2024-06-07:
@GiulioRomualdi on the latest `icub-firmware-build@devel` you should find the EMS firmware containing the fix, after your positive feedback I think we can close this issue (cc @AntonioConsilvio)

GiulioRomualdi, 2024-06-25:
I flashed https://github.com/robotology/icub-firmware-build/commit/10d4984af397f7b9d456cad317bee56d0b6cfa92 on the ergoCubSN002, I can test the fix

GiulioRomualdi, 2024-09-23:
We tested on ergoCubSN001. The robot behaved as expected

closed 2024-10-22 by AntonioConsilvio