robotology/mesh-support #1551: [iRonCub3] robot does not start

gabrielenava, 2023-05-04:
### Device name 🤖

iCubGenova09

### Request/Failure description

the robot does not start, both legs have errors on the hip joints boards.

### Detailed context

```
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 291m 443u: (code 0x00000018, par16 0x320d par64 0x000000000000021c) -> SYS: EOtheCANservice could not add a CAN frame in tx fifo. In par16 there is frame.id and frame.size (in most significant nibble). In par64 there is the frame.data. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 280m 428u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src CAN2, adr 0, time 130s 247m 122u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 247m 272u: (code 0x0000000b, par16 0x0217 par64 0x01d60217018701d6) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 294m 121u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src LOCAL, adr 0, time 130s 294m 284u: (code 0x0000000d, par16 0x0364 par64 0x4001036403ac031c) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 294m 442u: (code 0x00000018, par16 0x320d par64 0x000000000000021c) -> SYS: EOtheCANservice could not add a CAN frame in tx fifo. In par16 there is frame.id and frame.size (in most significant nibble). In par64 there is the frame.data. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 283m 428u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src CAN2, adr 0, time 130s 250m 122u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 250m 272u: (code 0x0000000b, par16 0x0217 par64 0x01d60217018601d6) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 297m 122u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src LOCAL, adr 0, time 130s 297m 285u: (code 0x0000000d, par16 0x0364 par64 0x4001036403ab031b) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 297m 443u: (code 0x00000018, par16 0x320d par64 0x000000000000021c) -> SYS: EOtheCANservice could not add a CAN frame in tx fifo. In par16 there is frame.id and frame.size (in most significant nibble). In par64 there is the frame.data. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 286m 427u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src CAN2, adr 0, time 130s 253m 122u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 253m 272u: (code 0x0000000b, par16 0x0217 par64 0x01d60217018701d7) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 300m 122u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src LOCAL, adr 0, time 130s 300m 285u: (code 0x0000000d, par16 0x0365 par64 0x4001036403ac031b) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 300m 442u: (code 0x00000018, par16 0x320d par64 0x000000000000021c) -> SYS: EOtheCANservice could not add a CAN frame in tx fifo. In par16 there is frame.id and frame.size (in most significant nibble). In par64 there is the frame.data. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 289m 428u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src CAN2, adr 0, time 130s 256m 121u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 256m 272u: (code 0x0000000b, par16 0x0217 par64 0x01d70217018601d6) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 303m 122u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
[WARNING]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src LOCAL, adr 0, time 130s 303m 285u: (code 0x0000000d, par16 0x0365 par64 0x4001036503ac031b) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
[ERROR]  from BOARD 10.0.1.11 (right_leg-eb11-j0_2), src CAN2, adr 0, time 130s 303m 443u: (code 0x00000018, par16 0x320d par64 0x000000000000021c) -> SYS: EOtheCANservice could not add a CAN frame in tx fifo. In par16 there is frame.id and frame.size (in most significant nibble). In par64 there is the frame.data. + .
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 130s 292m 427u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
```

AntonioConsilvio, 2023-05-04:
Hi @gabrielenava, the `yarprobotinterface` is looking for a CAN device on CAN2 port of the EMS boards in the hip. 
There should be FT58 sensors on CAN2, but they have been removed from the robot.

I removed the FT58 sensor from the iRonCub startup file, but in your log it looks like the `yarprobotinterface` is looking for something anyway.

After some testing, the problem seems to have been temporarily solved by commenting out the entire inertial sensor part within the iRonCub startup file.

I commented out the entire part because commenting out only the part that should call the IMU of FT58 is not sufficient.

However, this will not be the final solution, we need to understand why `yarprobotinterface` keeps looking for FT sensors.

cc @sgiraz @pattacini

sgiraz, 2023-05-05:
> After some testing, the problem seems to have been temporarily solved by commenting out the entire inertial sensor part within the iRonCub startup file.
> 
> I commented out the entire part because commenting out only the part that should call the IMU of FT58 is not sufficient.

cc @Nicogene @martinaxgloria

Nicogene, 2023-06-08:
> I commented out the entire part because commenting out only the part that should call the IMU of FT58 is not sufficient.

In theory it is required only to comment out [this line](https://github.com/ami-iit/robots-configuration/blob/8c0123fc82314ffdeea34386406427c40d0ee10c/iRonCub-Mk3/iRonCub_wbd.xml#L99) and [this line](https://github.com/ami-iit/robots-configuration/blob/8c0123fc82314ffdeea34386406427c40d0ee10c/iRonCub-Mk3/iRonCub_wbd.xml#L103), because these are xml files for running the imu service on strain2 boards

AntonioConsilvio, 2023-06-08:
Hi @Nicogene, thanks for the feedback!

I did it, but I don't know why it wasn't enough.

However, it will be tested again.

sgiraz, 2023-07-12:
Hi @gabrielenava,

Do you think we can close this issue?

gabrielenava, 2023-07-12:
I think yes, we did not test again [this problem](https://github.com/robotology/icub-tech-support/issues/1551#issuecomment-1535028345) but the robot can start with the current configuration, so I would close this issue

sgiraz, 2023-07-24:
Ok, we will proceed with closing this issue. If the problem persists or recurs in the future, we can reopen it or link it to a new one.
Thanks for your feedback!

cc  @AntonioConsilvio

closed 2023-07-24 by sgiraz