robotology/mesh-support #1552: [iRonCub3] problems with torso yaw joint

gabrielenava, 2023-05-04:
### Device name 🤖

iCubGenova09

### Request/Failure description

the torso yaw joint goes in hardware fault after some time, with the robot standing in position control.

### Detailed context

the error from yarpmotorgui is "hardware limits reached", but the robot is standing with the torso yaw in zero position:

```
[ERROR]  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 2948s 298m 251u: (code 0x02000010, par16 0x0002 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
```

maggia80, 2023-05-05:
this issue could be related to a wrong reading of the magnetic target. We might consider changing the calibration of this joint from absolute to incremental, but we need to check if an hw limit can be reached first.

sgiraz, 2023-06-16:
cc @AntonioConsilvio

AntonioConsilvio, 2023-06-22:
Hi @gabrielenava, has the problem recurred?

gabrielenava, 2023-06-23:
>Hi @gabrielenava, has the problem recurred?

not recently, we can close it and if it happens again during the experiments I will reopen it

closed 2023-06-23 by AntonioConsilvio