robotology/mesh-support #1554: iCubGenova11 neck_yaw sometimes goes into Hardware Fault

steb6, 2023-05-05:
### Device name 🤖

iCubGenova11

### Request/Failure description

iCubGenova11 neck_yaw sometimes goes into hardware fault.

### Detailed context

While using the iKinGazeCtrl and doing the usual HSP ergoCub demo, we noticed that the neck_yaw went into hardware fault when iCub tried to look down on its left.
With @AntonioConsilvio, @GiacomoVassallo13 and @sgiraz  we realized that when the neck moves when the neck_yaw is idle and almost on the limit.
We also realized that this could be due to the black cable in the photo below.
cc @xEnVrE 

### Additional context

[image]

### How does it affect you?

We may have problem running the ergoCub related demo with INAIL on May 8th.

GiacomoVassallo13, 2023-05-08:
Hello, the black sheath actually resisted the movement of the neck, thus increasing engine consumption by sending the system into protection. We experimented with various cord positions to find the one that offered the least resistance. After various attempts, the only solution that proved to be effective was to make the sheath come out not from the special exit flare on the cove but from the engine side as in the photograph. In this position the situation has definitely improved, and for now the problem does not seem to recur.

[image]

[image]

cc: @Fabrizio69 @AntonioConsilvio @sgiraz

Gandoo, 2023-05-11:
ciao, @sgiraz @AntonioConsilvio @GiacomoVassallo13 
 ho smontato le cover della testa e il supporto realsense, recuperato parte di cavo per allungare la corsa del collo e fare in modo che non sforzi il motore.
**Ho notato che la flex del motore centrale è allentata e per intervenire sul serraggio delle viti mi sa che bisogna staccare la testa e/o relativo telaio**

[image]

Facendo questo ho inserito la guaina nella sua sede della cover Zaino riportato come in origine. Fatto alcuni movimenti per verificarne la lunghezza.
Ricoverizzato la testa.

[image]

cc @xEnVrE @Fabrizio69 @maggia80

xEnVrE, 2023-05-11:
I tested this morning with one of our demos and it seems fine, thanks @Gandoo 

As regards the intervention you mentioned I would ask to postpone till May 17 as we are having important demos next Monday and Tuesday.

Thanks

sgiraz, 2023-05-11:
Ok @xEnVrE, We'll schedule the activity after that date.

@GiacomoVassallo13 @Fabrizio69

sgiraz, 2023-05-25:
@AntonioConsilvio, we may able to restart this activity as well, right?

xEnVrE, 2023-05-25:
> @AntonioConsilvio, we may able to restart this activity as well, right? 

Regarding this it would be good if we could coordinate with @fedeceola as it is going to use the robot quite intensively in the next weeks due to an incoming deadline.

xEnVrE, 2023-10-02:
I think this should stay open considering the intervention indicated in https://github.com/robotology/icub-tech-support/issues/1554#issuecomment-1543382001

AntonioConsilvio, 2023-11-23:
Hi @xEnVrE, let me know when you think the robot will be available to schedule this intervention.

Thank you!

xEnVrE, 2023-11-23:
Hi @AntonioConsilvio, I think that the user that might be most affected by this intervention is @fedeceola as he is planning experiments next days.

@fedeceola when do you think the robot can be available to fix this issue? Thank you

fedeceola, 2023-11-23:
Hi @AntonioConsilvio, @xEnVrE, if for you is fine and two days are enough for the intervention ideally I would say next Thursday and Friday (30th of Nov and 1st of Dec). Otherwise we can find another date. What do you think?

sgiraz, 2024-01-29:
Hi Guys, finally this issue has been solved by @GiacomoVassallo13 and then tested by @martinaxgloria and @AntonioConsilvio.
Well done guys! 🚀
Closing as solved!

cc @fedeceola @steb6

closed 2024-01-29 by sgiraz