robotology/mesh-support #1555: Left thumb adduction not moving

S-Dafarra, 2023-05-08:
### Device name 🤖

ergoCub S/N000

### Request/Failure description

The left thumb adduction does not move. It makes a weird noise, like a spinning motor and it can be moved by hand, but it cannot be controlled

### Detailed context

We were performing a teleoperation demo and that joint might have been stressed while trying to hold a box

AntonioConsilvio, 2023-05-11:
Hello @S-Dafarra, I fixed the problem by changing some values in the configuration files and the cable that connect the FAP sensor to the MMA board.

A PR to update robots-configurations upstream is coming soon.

cc @sgiraz

sgiraz, 2023-05-15:
PR open here:
- https://github.com/robotology/robots-configuration/pull/512

closed 2023-05-16 by AntonioConsilvio