robotology/mesh-support #1557: Left arm FT sensor not working

S-Dafarra, 2023-05-08:
### Device name 🤖

ergoCub S/N000

### Request/Failure description

The left arm FT sensor reads only zeros

### Detailed context

The robot is able to start, but then the ``yarprobotinterface`` complains about the fact that a CAN sensor on the left arm does not start streaming. The readings from the corresponding yarp port are all null.

AntonioConsilvio, 2023-05-11:
Hi @S-Dafarra, the problem should be fixed by checking the wiring of the FT sensor (name of the cable: 2n2).

However, give us a feedback about it.

cc @sgiraz

sgiraz, 2023-05-17:
I put this issue on hold for a while, as it has not happened anymore.

sgiraz, 2023-06-05:
Closing as it doesn't seem to have happened again. Feel free to re-open if necessary.

S-Dafarra, 2023-06-08:
> During the conference, the robot went in protection once (motors LED blinking) and checking the log there seemed to be a lot of errors with the CAN [log_ergocub_yarprobotinterface_robot_fall.zip](https://github.com/robotology/icub-tech-support/files/11653582/log_ergocub_yarprobotinterface_robot_fall.zip)
> 
> In any case, now the problem has moved to the right arm FT, while on the left some CRC errors occurred

cc @pattacini @maggia80

paolo-viceconte, 2023-06-09:
We are having the same problem right now while starting the robot. 

Here the logger exported on file:

GiulioRomualdi, 2023-06-12:
As a workaround, we removed the FT from 
1. Yarprobotinterface (both hardware and wrapper)
2. Whole body dynamics
3. The urdf model

In this way, the robot starts. Still, it is not a desiderable configuration.

AntonioConsilvio, 2023-06-15:
Hi @S-Dafarra @GiulioRomualdi, yesterday me and @Gandoo solved this problem. We added a CAN line termination, so the problem should no longer occur. 

However, I know there are several demo today, so give us feedback after these demo.

cc @sgiraz @Fabrizio69

Gandoo, 2023-06-16:
On the CAN added this Y to put the 120 ohm resistor

[image]

cc @AntonioConsilvio @sgiraz @Fabrizio69 @maggia80

sgiraz, 2023-06-16:
Ok the problem related to the FT sensor should be solved.
I'm going to close this issue.

Thanks @Gandoo and @AntonioConsilvio!

closed 2023-06-16 by sgiraz

S-Dafarra, 2023-06-16:
Apologies, I did not manage to answer before. The FTs are working fine.