robotology/mesh-support #1559: Connect speakers and microphone to the same board

S-Dafarra, 2023-05-11:
### Device name 🤖

ergoCub S/N000

### Request/Failure description

At the moment, the speaker is connected to the ergocub head, while the microphone to the torso. Both are working well, but the issue is that being in two separate computers, it is not possible to run echo cancellation. This is a problem when using both the microphone and the speaker at the same time. Everything that comes out of the speakers is recorded by the microphone, usually with a volume much higher than any other sounds in the enviroment.

### Detailed context

When we run the teleoperation, the voice of the operator is replayed on the speaker, but then the operator hears his voice back with some delay because of this echo effect. In the past we managed to run some echo cancelling software, but this requires the speaker and the microphone to be connected to the same board.

S-Dafarra, 2023-05-11:
cc @maggia80 @GiulioRomualdi @mebbaid @lrapetti

sgiraz, 2023-05-11:
Hi @S-Dafarra, 

Great observation!
I have the following suggestions for this issue: 
1. Start by reducing the volume of the speaker, so that the sound picked up by the microphone is not too loud.
2. Adjust the microphone sensitivity, if possibile (via SW/HW)
3. Use software tools that can help reduce echo, such as echo cancellation software or acoustic echo cancellation (AEC) algorithms.

> **Warning**
> Anyway, this is not considered a support issue, but rather a feature request. To get more feedback, I recommend opening a ticket.

cc @maggia80 @Fabrizio69 @MrAndrea

S-Dafarra, 2023-05-11:
> * Start by reducing the volume of the speaker, so that the sound picked up by the microphone is not too loud.

Unfortunately, this is not viable since the volume is not so high. In addition, being the speaker close to the microphone, the volume would need to be almost to zero, but at that point it would be useless. The speaker is the one thing that the microphone picks up best.

> 2\. Adjust the microphone sensitivity, if possibile (via SW/HW).

The microphone sensitivity is already low and I had to increase all the gains to maximum to make a person speaking in front of the robot audible. Hence, neither this solution is viable.

> 3\. Use software tools that can help reduce echo, such as echo cancellation software or acoustic echo cancellation (AEC) algorithms.

Yes, we have experience with that, but they work by accessing the mic and the speaker at the same time. Clearly this is possible only if they are connected to the same PC.

> Warning
> Anyway, this is not considered a support issue, but rather a feature request. To get more feedback, I recommend opening a ticket.

Thanks for pointing that out. I was a bit in doubt, but since this is a sort of underperformance that requires intervening on the robot, I thought this was a better place. I can move it if you prefer.

MrAndrea, 2023-05-11:
Unfortunately the USB-audio adapter that we added in the head can manage only a mono microphone, that's why we didn't change the connection of the 2 microphones.
The "quicker" solution is to connect only one microphone to that adapter (if you don't need stereo microphones), then we can look for a different adapter or another to add.

MrAndrea, 2023-05-11:
Something like this: [RØDE AI-Micro](https://rode.com/it/interfaces-and-mixers/ai-series/ai-micro)

[image]

(on [Amazon](https://www.amazon.it/AI-Micro-uninterfaccia-ultra-compatta-registrare-dispositivo/dp/B09KZJTW73/ref=asc_df_B09KZJTW73/?tag=googshopit-21&linkCode=df0&hvadid=555651669282&hvpos=&hvnetw=g&hvrand=6448015483951730464&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1008337&hvtargid=pla-1508937527160&th=1))

MrAndrea, 2023-05-12:
> I think that would be double. Indeed, at the moment we stream only mono, so I think it should not be a big loss.

In this case, we need a 4-pole 3.5mm jack like [this](https://it.rs-online.com/web/p/connettori-jack/8051652) (see also this [comment](https://github.com/icub-tech-iit/project-ergocub/issues/83#issuecomment-1514863085)), or a pre-assembled cable like the following:
- [Cavo assemblato audio/video, jack maschio a 4 poli 3,5mm, estremità libera, 6.6 ft, 2 m, nero](https://it.farnell.com/cliff-electronic-components/fc68122c/assembl-cavi-spina-jack-4-vie/dp/2470995?gclid=Cj0KCQjwpPKiBhDvARIsACn-gzDSdRQ0jj0CynaeLYLn4blQffvKGEG2kmoH8GCRRYLLWTOYGdKS9SEaAmibEALw_wcB&mckv=_dc|pcrid||plid||kword||match||slid||product|2470995|pgrid||ptaid|&CMP=KNC-GIT-GEN-SHOPPING-PMAX-Med-Roas-Short-Title-Test-21-Dec-22&gross_price=true)
- [Cavi assemblato audio/video, jack maschio a 4 poli 3,5mm, jack maschio a 4 poli da 3,5mm](https://it.farnell.com/w/c/cavi-fili-e-assemblati/cavi-assemblati/cavi-assemblati-audio-video?connettore-tipo-b=jack-maschio-a-4-poli-da-3-5mm&st=cavo%20assemblato%20jack%20maschio%204%20poli%203%2C5mm)

[image]

[image]

closed 2023-05-12 by sgiraz