robotology/mesh-support #1560: Loosy left elbow

DanielePucci, 2023-05-12:
### Device name 🤖

ergoCub

### Request/Failure description

https://github.com/robotology/icub-tech-support/assets/5045846/90da2003-ef5c-46d1-bdcd-f376a97927b0

### Detailed context

none

sgiraz, 2023-05-15:
### On the elbow

The 3 screws indicated below have been tightened:

[image]

--- 

Moreover, @Gandoo and @AntonioConsilvio spotted that also the following screws were loose, so they re-tightened them.

[image]

After these maintenance activities, the issue should be solved.
We are waiting for your feedback.

cc @DanielePucci

sgiraz, 2023-05-17:
Closing! Feel free to reopen if necessary.

cc @AntonioConsilvio

closed 2023-05-17 by sgiraz