robotology/mesh-support #1562: R1SN003 base pc CMOS battery substitution

elandini84, 2023-05-16:
### Device name 🤖

R1SN003

### Request/Failure description

The CMOS battery of the base PC of R1SN003 appears to be dead since the BIOS configuration gets reset every time the pc is turned off and back on.

### Detailed context

When we turned on the robot yesterday (15/05/2023) we found out that on the base pc no boot partition could be found. After further investigation we realized that BIOS configuration was back to the default one.
We tried to reconfigure the BIOS and the pc booted up as expected. When we then tried to turn it off and on again we got the same exact situation (default BIOS configuration).

randaz81, 2023-06-08:
I'm not sure about what a sprint is exactly. Please provide an estimated date...
Please also consider that the issue has been opened already 3 weeks ago, we are now working with some difficulty and we have demos in the upcoming week (14-15th June) :-(
Thank you for your understanding

sgiraz, 2023-06-09:
Hi @randaz81, the maintenance activity has been scheduled for Monday 9/06 

cc @fbiggi @Gandoo @Fabrizio69

randaz81, 2023-06-09:
do you mean Monday 12/06?

sgiraz, 2023-06-09:
> do you mean Monday 12/06?

Yep! Monday 12/06, my bad.

fbiggi, 2023-06-12:
@AntonioConsilvio @sgiraz batteria sostituita 
inoltro la [issue](https://github.com/icub-tech-iit/proto/issues/2311) dell'intervento

AntonioConsilvio, 2023-06-12:
Hi @elandini84 @randaz81, today @fbiggi and @Gandoo have replaced the robot's CMOS battery. 

Please give us feedback when you use the robot.

cc @sgiraz @Fabrizio69

elandini84, 2023-06-12:
> Hi @elandini84 @randaz81, today @fbiggi and @Gandoo have replaced the robot's CMOS battery.
> 
> Please give us feedback when you use the robot.
> 
> cc @sgiraz @Fabrizio69

Everything works as expected. Thanks a lot.

sgiraz, 2023-06-12:
Well, done guys! Closing.

closed 2023-06-12 by sgiraz