robotology/mesh-support #1563: Broken Cable in the Index & Missing Skin on the Thumb and Middle Finger (Left Hand)

Matilde-Antonj, 2023-05-16:
### Device name 🤖

iCubGenova03

### Request/Failure description

It should be necessary to replace the broken cable in the index of the left hand and attach new patches of skin on the tip of the thumb and middle finger of the left hand.

### Detailed context

**Index of the left hand:**
The index of the left hand is continuously bended. When you try to straighten it, you can feel an opposite force that doesn't allow you to do it. Simultaneously, you hear the sound of the motor that tries to move the finger in the right location. It seems to be a broken cable in the index.

**Skin of the thumb and middle finger in the left hand:**
The skin, applied on the tip of the thumb and the middle finger in the left hand, was removed, therefore these parts of the hand need new patches of skin to apply there.

### Additional context

[image]

https://github.com/robotology/icub-tech-support/assets/103580025/fe61a941-b343-4235-a28b-96a895efa3fd

sgiraz, 2023-06-13:
Hi @Matilde-Antonj,

Maintenance activity scheduled for today. @AntonioConsilvio And @GiacomoVassallo13 are going to reach you this afternoon to fix the issues mentioned above.

Cc @Fabrizio69

AntonioConsilvio, 2023-06-14:
Hi @Matilde-Antonj, We found several problems during the intervention:

- A piece of the damaged cable got stuck inside the phalanx  
- The gearbox was split from the motor (left proximal index)
- The MAIS board was detached
- The cable of the sensor of the left wrist was detached.

Despite these problems we replaced the thumb and ring fingertips (we replaced the ring fingertip because it was without the skin, not the middle). Then, replaced the tendon by making the knot at the base of the phalanx and it works, but it do a bad sound, so maybe we will replace the motor:

https://github.com/robotology/icub-tech-support/assets/114915464/f1f96189-c4e7-4634-961f-fc8d498930cf

So the robot is usable.

cc @sgiraz

Matilde-Antonj, 2023-06-15:
Hi @AntonioConsilvio,
Thank you for this report!
Let me know when you come back in the lab to replace the motor.
Have a nice day!

cc @sgiraz

sgiraz, 2023-09-29:
Hi @Matilde-Antonj,

We are going to Erzelli on next Monday (2nd of October), if you agree we can try to conclude the maintenance activity on site.

cc @AntonioConsilvio @GiacomoVassallo13

Matilde-Antonj, 2023-09-29:
Hi @sgiraz,
That's perfect.
Thank you!

cc @AntonioConsilvio @GiacomoVassallo13

AntonioConsilvio, 2023-10-03:
Yesterday we came to Erzelli to perform two interventions, the one concerning this issue was the second and unfortunately due to timing reasons we were unable to perform it, so it is necessary to reschedule the intervention.

Fortunately for now the problem is not blocking, so the robot can be used to perform the experiments

cc @sgiraz @Fabrizio69 @Matilde-Antonj

Matilde-Antonj, 2023-10-03:
Hi @AntonioConsilvio,
Thank you for your report!
Can we schedule the next intervention?
Thank you very much!

cc @sgiraz @Fabrizio69

AntonioConsilvio, 2023-10-05:
Hi @Matilde-Antonj, sorry, unfortunately this will be a period with several deadlines for us, so we cannot yet confirm a date for the intervention.

The issue will be updated as soon as we can confirm our availability.

cc @sgiraz

Matilde-Antonj, 2023-10-05:
Hi @AntonioConsilvio,
I will wait for updates on the date for the intervention.
Thank you!

cc @sgiraz

sgiraz, 2023-10-16:
Hi @Matilde-Antonj,

We are planning the activity for the second week of November (6th of November).
Let me know if it is a good date for you.

cc @AntonioConsilvio @GiacomoVassallo13

Matilde-Antonj, 2023-10-17:
Hi @sgiraz,
It is ok for us!
Will you come in the morning or in the afternoon?
Thank you very much!
See you on 6th November.

cc @reafrancesco @AntonioConsilvio @GiacomoVassallo13

sgiraz, 2023-10-19:
> Will you come in the morning or in the afternoon?

Hi @Matilde-Antonj,
I can confirm that We will be on-site early in the morning.

cc @AntonioConsilvio @GiacomoVassallo13

AntonioConsilvio, 2023-11-07:
Hi @Matilde-Antonj, yesterday me and @GiacomoVassallo13 replaced the phalanx and the damaged motor. 
During the intervention, we also replaced two hall sensors in the fingers and remade the proximal and distal tendon of the index finger.

The intervention ended with restoration of full function of the proximal index finger.

Unfortunately, during the testing phase we also found a malfunction on the distal of the index finger that requires further intervention.

However, this issue has been solved, another issue will be opened for the problem on the index distal.

Closing!

cc @sgiraz

closed 2023-11-07 by AntonioConsilvio

Matilde-Antonj, 2023-11-07:
Hi @AntonioConsilvio and @GiacomoVassallo13,
Thank you very much for your work and your time.
I will wait to be tagged in the next issue, in order to be updated for the other intervention.
Have a nice day!

sgiraz, 2023-11-07:
> Unfortunately, during the testing phase we also found a malfunction on the distal of the index finger that requires further intervention.

just for reference:
- https://github.com/robotology/icub-tech-support/issues/1674

AntonioConsilvio, 2023-11-10:
Hi @Matilde-Antonj  @MissingSignal, sorry for the somewhat late information, I wanted to inform you that during the intervention we had to disassemble and reassemble the magnets and this operation caused the need to reverse the PID of the index distal joint.

The PR to update the robot-configurations files is here:
- https://github.com/robotology/robots-configuration/pull/588