robotology/mesh-support #1564: check for R1SN001 main battery replacement

randaz81, 2023-05-16:
### Device name 🤖

R1SN001 Erzelli

### Request/Failure description

Pavis research team notified us that the battery charge of R1SN001 seems to be unexpectedly short. 
Currently, we do not have a quantitative measurement. 
However, this might be a trigger for us to check if we already have a replacement battery or if we need to order components (and their lead time)

### Detailed context

---

maggia80, 2023-05-17:
@randaz81 we are going to open a purchase order for a new battery.

maggia80, 2023-09-19:
@randaz81 we have issues in purchasing the replacement since the company is not producing anymore this battery pack. I'm searching for an alternative solution/provider

AntonioConsilvio, 2024-08-06:
Hi @randaz81! The battery has arrived, we can proceed with the replacement.

We were planning to do this on Thursday 8th August.

To get to the robot, do the replacement and test the battery, we are autonomous.
But to test the robot's functionality, we need someone to start it.

Is someone available on Thursday?

cc @elandini84

randaz81, 2024-08-07:
Top! Lets's do it!

AntonioConsilvio, 2024-08-09:
Hi @randaz81, yesterday me and @fbiggi replaced the battery. The battery is working and the robot is powered correctly. 

Please give me feedback on the functioning of the robot when you test it!

fbrand-new, 2024-08-20:
We tested the robot and can confirm that it is working correctly, thank you!

closed 2024-08-20 by fbrand-new