robotology/mesh-support #1565: ergoCubSN000 - left wrist does not move

steb6, 2023-05-18:
### Device name 🤖

ergoCubSN000

### Request/Failure description

While testing the ICRA 2023 demo we noticed that the left wrist was a bit loose and then we  realized that it was not possible to move it anymore sending control commands.

### Detailed context

 @Woolfrey @xEnVrE @andrearosasco @lrapetti

sgiraz, 2023-05-18:
Hi @steb6,

Tomorrow morning we gonna check the wrist's screws and replace the (already spotted by @AntonioConsilvio) AEA faulted wire.

cc @GiacomoVassallo13 @Fabrizio69 @Gandoo

Gandoo, 2023-05-19:
ciao @AntonioConsilvio @sgiraz  il supporto sempra abbia subito qualche danno

https://github.com/robotology/icub-tech-support/assets/53341641/2b05de58-51d2-462d-a315-a427e043bb12

cc @maggia80 @Fabrizio69

sgiraz, 2023-05-19:
@Gandoo and @AntonioConsilvio removed the cover from the forearm and spotted that the AEA support was **broken**!