robotology/mesh-support #1567: iCubGenova11 - Power cable is scratching the robot cover and causing shutdowns

xEnVrE, 2023-05-22:
### Device name 🤖

iCubGenova11

### Request/Failure description

While doing experiments @fedeceola noticed that the power cable is going against the robot cover in certain configurations.

For example, when the torso is configured as follows:

[image]

the plug connector is that near to the cover:

[image]

@fedeceola also notified that while using the robot it happened that the robot passed by that configuration and a complete shutdown happened - presumably because the connector was hardly going against the cover.

### Detailed context

It seems beneficial to maybe revise the limits of the torso to avoid such situations?

cc @sgiraz

AntonioConsilvio, 2023-05-23:
Hi @xEnVrE, this morning @Gandoo eliminated the angle of the connector. 

[image]

Now we should find out whether it is sufficient or not.

cc @sgiraz

xEnVrE, 2023-05-24:
Today it happened again while we were making a video. When the shutdown happens I can hear a sort of "click":

https://github.com/robotology/icub-tech-support/assets/6014499/1f1c7712-56a2-4e5b-bff5-30b3a57ca11e

What is strange to me is that when the shutdown happens the robot is not moving the torso. Maybe the connector was already going against the cover and some safety mechanism - e.g. based on current sensing - removed the power? Can you confirm whether the battery pack has any protection mechanism?

I want to remark the fact that after the shutdown the two LEDs on the battery pack were RED (as if the main power was turned off and then on again) while the power amplifier was on as usual.

cc @sgiraz @AntonioConsilvio

sgiraz, 2023-05-25:
cc @maggia80 @MrAndrea @Gandoo

AntonioConsilvio, 2023-05-26:
Hi @xEnVrE with the help of @maggia80, we found the a damaged power cable in the waist:
[image]

I eliminated the burnt part of the cable, and soldered the two ends of the cable.
Since we decided not to keep the waist cover to avoid the collision between the power connector and the cover, I arranged the cables so that they are not too loose during robot movement.

[image]

cc @sgiraz

MrAndrea, 2023-05-26:
Hi @AntonioConsilvio, I suspect that after the [short circuit happened](https://github.com/robotology/icub-tech-support/issues/1567#issuecomment-1564118974) some components on the BDC board could be damaged. When a push button blinks, there is an overcurrent as explained in this [wiki page](https://github.com/icub-tech-iit/electronics-boards/wiki/power-push-buttons), but I don't think is an overcurrent because at the start-up, the regulators and the HSM are disabled by the BCB board, and both the 2 push button should be red.
Is the red cable the "48V" line? If yes, probably there is something damaged in the HSM circuit on the BDC board (and cascading may be on the other regulators as well).
My suggestion is to replace for the time being only the BDC board and see if the system works fine.

xEnVrE, 2023-05-26:
Thanks for the information @MrAndrea.

> there is an overcurrent as explained in this [wiki page](https://github.com/icub-tech-iit/electronics-boards/wiki/power-push-buttons)

btw, we were searching for this kind of information in the iCubTech public documentation but we could not find it. Should not it be available to everyone? Thank you

MrAndrea, 2023-05-26:
@xEnVrE currently there isn't a public page, I will add it in the next days on documentation.

MrAndrea, 2023-06-05:
Hi @xEnVrE, here is the public page:

https://icub-tech-iit.github.io/documentation/robot_power/robot_power_button/

sgiraz, 2023-06-07:
Since the power connector scratching and shutdown problem have been (partially) solved,  we'll open an internal issue to address the `BDC` board replacement.

cc @AntonioConsilvio