robotology/mesh-support #1568: iCubGenova11 - The neck encoders do not read

xEnVrE, 2023-05-24:
### Device name 🤖

iCubGenova11

### Request/Failure description

The encoders of the neck do not read at all.

### Detailed context

@AntonioConsilvio verified that the issue is due to the SPI interfaces of two MC4-PLUS not working.

### Additional context

The issue has been already fixed by @AntonioConsilvio.

Please feel free to add more details not the problem and its solution. Thanks

AntonioConsilvio, 2023-05-26:
Hi @xEnVrE, I discovered that the SPI connectors of the MC4-PLUS EB20 did not give 5 volt to power the AEA sensors. 

So I replaced the board with a new one and teste. 

Now the robot works properly!

cc @sgiraz

xEnVrE, 2023-05-26:
Thanks @AntonioConsilvio

closed 2023-05-29 by sgiraz