robotology/mesh-support #1570: [iCubErzelli02] Middle finger tendon damaged (right arm)

ddetommaso, 2023-05-29:
### Device name 🤖

iCubErzelli02

### Request/Failure description

one of the tendon seems damaged and out of its pulley

### Detailed context

-

### Additional context

[log_icub-head_yarprobotinterface_2992.log](https://github.com/robotology/icub-tech-support/files/11590436/log_icub-head_yarprobotinterface_2992.log)

### How does it affect you?

Please we would ask you to intervene promptly because we have participants already scheduled in the next days for running experiments that need to be cancelled if the robot does not work properly.

sgiraz, 2023-05-30:
Hi @ddetommaso,

We received your support request!
if available, I would ask you to drop here some pictures o videos in order to see if there are other problems around the same joint.

Cc @Fabrizio69 @AntonioConsilvio @GiacomoVassallo13 @fbiggi

sgiraz, 2023-06-05:
Hi @GiacomoVassallo13,

Can you share some pictures that illustrate the results of the maintenance work carried out this morning?
If I understood correctly from our face-to-face call, there are no longer any mechanical problems, but the robot is still experiencing calibration issues possibly due to wiring problems, is that correct?

cc @AntonioConsilvio @Fabrizio69

sgiraz, 2023-06-05:
Thanks @ddetommaso, looking at the first ERROR:

```console
 29.774677 <ERROR>  from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 92s 965m 395u: (code 0x05000016, par16 0x4006 par64 0xfffffffff0000110) -> CFG: EOtheEncodeerReader cannot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
```

we can see that there may be a wiring problem with the `mc4plus` with IP `10.0.1.27` and the encoders connected to it.

cc @AntonioConsilvio

AntonioConsilvio, 2023-06-08:
Hi @ddetommaso, after the middle tendon replacement by @GiacomoVassallo13 , yesterday (7th June) I fixed the following problems:

- [X] Eliminated a short circuit in the right MTB4 forearm (CAN line of the EB27 board)
- [X] Unlocked the motors of the pinky and middle finger as they were mechanically blocked by the plates covering them.
- [X] Replaced blown CAN cable (4N1_4)
- [X] Rewired cables RA_2S2 and RA_2S1, which connect the wrist sensors to the EB27 board.

However, I identified the following problems that require further work:

- [ ] The distal of the thumb closes, but does not open (can be opened by hand)

- [ ] Right shoulder joint 0 calibration not accurate (non-locking problem)

- [ ] Problems on the CAN line of the EB10 board.

- [ ] Maybe there is a problem on the left hand pinky sensor reading.

The robot has been tested and is able to perform the demo and experiments for which it is required at this time.

However, even if these problems do not impede this specific demo from running, it is best not to leave them hanging for too long.

cc @Fabrizio69 @sgiraz

ddetommaso, 2023-06-08:
Hi @AntonioConsilvio,

many thanks for the intervention and the detailed feedback.
If you guys agree I would keep open the issue until we finalize the intervention for the remaining problems.

We'll continue testing once the data collection is done and then we'll ask again your support.

Thanks

AntonioConsilvio, 2023-07-03:
Hi @ddetommaso , on Friday 30th me and @GiacomoVassallo13 came to Erzelli and we have:

- solved this [issue](https://github.com/robotology/icub-tech-support/issues/1583) 
- Replaced sheath and tendon in the distal of the right thumb
- Replaced the tendon of the right wrist because it was torn
- Repaired almost all the MTB wiring (skin) on the left leg as it was damaged
- Resoldered two fingertips of the right hand (thumb and middle finger)
- Resoldered the skin of the right palm
- Fixed the 4N2_7 cable because it was damaged (right forearm skin) 
- Resoldered AEA5 sensor cable (RA_2S1) right wrist
- Replaced the holder of the AEA5 because it was damaged
- Replaced AEA1 sensor cable (RA_0S1_a) right shoulder
- Tested the left hand pinky and it was working properly

Unfortunately rewiring the left leg skin did not solve the problem on the EB10's CAN, as the problem seems to be with the EB10 board itself. It was not replaced as more urgent problems were found during maintenance of the right arm.
The recalibration of the right shoulder was also not done due to more urgent problems. 

The robot has been tested and all joints work correctly, the robot is usable. 

cc @Fabrizio69

ddetommaso, 2023-07-04:
Hi @AntonioConsilvio,

thank you and @GiacomoVassallo13 for the great job!
We're still having some issue on the left arm because the l_pinky seems not working correctly (in the yarpmotorgui is shown as NOT CONFIGURED).

sgiraz, 2023-07-05:
> We're still having some issue on the left arm because the l_pinky seems not working correctly (in the yarpmotorgui is shown as NOT CONFIGURED).

Hi @ddetommaso, 
I recommend opening a new issue specifically for the problem mentioned earlier, as the current one should focus on the right arm. Of course, when we perform maintenance on this robot, we will thoroughly inspect and repair any existing faults, if feasible.

ddetommaso, 2023-07-08:
Hi @sgiraz,

I see your point but I was referring to one of the faults mention above still in this thread:

> * Maybe there is a problem on the left hand pinky sensor reading.

and

> * Tested the left hand pinky and it was working properly

sgiraz, 2023-07-10:
### Maintenance activity report 10/07/23

**Interventions:**
| Description | pictures | Note |
| :---: | :---: | :---: |
| re-soldered left elbow AEA wires `L1S1` | ![AEA wires][1] | 2 wires were broken ⚫🟡 |
| replaced a screw M1x3 for AEA pcb on elbow  | ![AEA screw][2] | there was a too long screw mounted for that pcb |
| added joint 15 in `left_arm` calibrator confi files | | it was missing probably due to some tests |

**Tests:**
| Tests | pictures | Note |
| :---: | :---: | :---: |
| right arm joints | | |
| left arm joints | | |
| whole robot | | |
| demo to be performed next Wednesday | | |

**Notes:**
- neck tendons loosen
- tendon of `l_pinky` coupled joint needs to be improved (little finger close earlier than annular). Moreover, the tendon may be too long.
- right-hand adduction is too loose compared to the left one.
- friction on the right-hand annular that makes the joint movement not smooth
- `eb10` CAN board issues are still present. It may be possible to remove the boards as the lower body is not used at all.

cc @Fabrizio69 

[1]: https://github.com/robotology/icub-tech-support/assets/18039045/1b773ad8-3123-4425-83b5-ae52ad78168c
[2]: https://github.com/robotology/icub-tech-support/assets/18039045/96d3fb79-7976-46cc-876f-d0d1fc7de8dd

sgiraz, 2023-07-12:
Hi @ddetommaso,

I hope the demos of today went well and if you agree I'm going to close this and the #1592 tomorrow.

ddetommaso, 2023-07-13:
Hi @sgiraz 

the demo went well, thank you very much for all your work.

As side note I add that just after the demo the robot shut down without any specific reason. I know that this robot had similar issue in the past #1315 but the cause was never found.  Could it be related to the problem of the legs in fault?

I think we can close this issue but it would be good to keep open separate threads for the issues that still need to be solved or need further investigations. Please collect all these issues that you found in these days and keep them alive the way you prefer. Maybe a check in the history of this robot could help https://github.com/robotology/icub-tech-support/issues?q=iCubErzelli02

Thank you

closed 2023-07-13 by ddetommaso

sgiraz, 2023-07-24:
Ok, we are planning to schedule a full check maintenenance activity for `iCubErzelli02,` `iCubErzelli01` and `iCubGenova03` (one robot at once), starting from the week after 15th of August. 

cc @reafrancesco @AntonioConsilvio @maggia80 @Fabrizio69

sgiraz, 2023-07-26:
/remind 16 Aug Organize the transport of iCubErzelli02 from Erzelli to CRIS (cc @Fabrizio69 @AntonioConsilvio @maggia80)