robotology/mesh-support #1573: JOINT 6(I_wrist_yaw) - HARDWARE FAULT

jimingda, 2023-06-06:
### Device name 🤖

iCubShanghai01

### Request/Failure description

The message on the yarpmotorgui about left_arm is as below:

```
JOINT 6(i_wrist_yaw)-HARDWARE FAULT
Fault Code: 16
FaultMessage: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID

Encoder: 45.434 deg
```

How do i fix this fault?

Regards

### Detailed context

None

jimingda, 2023-06-07:
Actually today when i turn on icub, and launch yarprobotinterface again, the left arm go back to normal again as well as the joint 6.
All the status of joints on yarpmotorgui is green now.

sgiraz, 2023-06-16:
I guess the problem is gone, hence I'm going to close this issue.
Feel free to reopen in necessary.

closed 2023-06-16 by sgiraz