robotology/mesh-support #1575: iCubGenova11 - Left hand joints start in idle

xEnVrE, 2023-06-07:
### Device name 🤖

iCubGenova11

### Request/Failure description

This morning we noticed that some joints of the left hand start in idle when the `yarprobotinterface` is executed.

[image]

### Detailed context

Attaching the log file.

[log_icub-head_yarprobotinterface_1938.txt](https://github.com/robotology/icub-tech-support/files/11675712/log_icub-head_yarprobotinterface_1938.txt)

### How does it affect you?

We have several demos next week, it would be good - if possible - to investigate this and understand why this happens. 

In the meanwhile, would it be safe to **run** the joints starting in idle, i.e. by pressing `run` on the `yarpmotorgui`?

Thanks

cc @sgiraz @AntonioConsilvio

sgiraz, 2023-06-16:
Hi @AntonioConsilvio,

As you noticed last time, it seems that this joint may need to be recalibrated. Please keep @xEnVrE informed, as he has modified the high-level software code to allow the joint to be used regardless.

sgiraz, 2023-07-04:
Also, we should check for unsoldered cables on the `Mais` boards of the left hand

pattacini, 2023-07-04:
This should be solved now.
cc @martinaxgloria

sgiraz, 2023-07-05:
Hi @pattacini,

After a f2f meet-up with @martinaxgloria and @AntonioConsilvio, it seems to be a random issue. 
Hence there are 2 possibilities:
1. leave everything as is and re-open this issue whenever the problem occurs again
2. double check the `Mais` boards as mentioned above and look for bad contacts 

cc @Nicogene

xEnVrE, 2023-07-05:
Hi @sgiraz, should we move the additional problem indicated in https://github.com/robotology/icub-tech-support/issues/1575#issuecomment-1589494200 to a separate issue? I feel that the two problems are maybe related but still different and I am afraid the latter might be lost in this issue.

sgiraz, 2023-07-06:
> Hi @sgiraz, should we move the additional problem indicated in [#1575 (comment)](https://github.com/robotology/icub-tech-support/issues/1575#issuecomment-1589494200) to a separate issue?

Yep! Feel free to open a separate issue.

AntonioConsilvio, 2023-08-04:
Hi @xEnVrE, yes, the two problems were related, in fact this [intervention](https://github.com/robotology/icub-tech-support/issues/1593#issuecomment-1664224433) solved this issue as well.

The problem was that the distal of the left thumb didn't calibrate correctly (because even if it was all open, the encoder read 90 degrees instead of 0) and therefore all the distal joints went into idle, because by calibrating together, if one of them fails, the others go into idle accordingly.

So, the issue should be solved, but if the problem reappears, feel free to reopen the issue.

cc @sgiraz @martinaxgloria

closed 2023-08-04 by AntonioConsilvio