robotology/mesh-support #1576: what is the recommanded default gesture of icub when not in use

jimingda, 2023-06-13:
### Device name 🤖

iCubShanghai01

### Request/Failure description

As default, if properly exit the application in yarpmanager,  the head and torso of icub are tilted, as the gesture in the pic below rather than upright, is it the recommand situation？

### Detailed context

[image]

jimingda, 2023-06-13:
The reason we have this question is that it doesn't looks normal, but maybe this tilted situation is good for icub, can reduce the tension on the cable driven, and extend its service life.

AntonioConsilvio, 2023-06-15:
Hi @jimingda, the position the robot assumes once the yarprobotinterface is closed is completely normal. 

If in the first period the robot's neck remained straight instead of falling, it was simply due to the friction produced by the gearbox.

After a bit of running in, the gearbox produces less friction, so it's actually a positive situation.

So, there are no problems about it!

cc @sgiraz

sgiraz, 2023-06-16:
Hi @jimingda,

The answer provided by @AntonioConsilvio should have clarified your doubts, so I am going to close this issue. Feel free to reopen it if necessary.

closed 2023-06-16 by sgiraz