robotology/mesh-support #1577: iCubGenova11 - Right arm r_elbow AEA not working

xEnVrE, 2023-06-13:
### Device name 🤖

iCubGenova11

### Request/Failure description

The right arm `r_elbow` encoder is not working as expected.

### Detailed context

[image]

### Additional context

Please find here the log of the `yarprobotinteface`:

[log_icub-head_yarprobotinterface_3943.txt](https://github.com/robotology/icub-tech-support/files/11736389/log_icub-head_yarprobotinterface_3943.txt)

### How does it affect you?

We would need to repair this before Wednesday afternoon due to some demos on Thurdsday morning and afternoon. Thank you

cc @sgiraz @AntonioConsilvio @AntonioAzocar

AntonioConsilvio, 2023-06-15:
Hi @xEnVrE, yesterday I found the cable R1S1, that connect the AEA sensor to the EMS EB3, broken under the shoulder tendon:
 
[image]

I and @Gandoo replaced the cable and tested the joint and it works.

https://github.com/robotology/icub-tech-support/assets/114915464/a0c31d33-4bea-4360-b0f8-83fd894437cb

Feel free to reopen the issue if the problem reappears.

cc @sgiraz

closed 2023-06-15 by AntonioConsilvio

xEnVrE, 2023-06-15:
Thanks a lot!! @AntonioConsilvio @Gandoo @sgiraz