robotology/mesh-support #1578: ergoCub SN000 hands fail to calibrate

S-Dafarra, 2023-06-14:
### Device name 🤖

ergoCub S/N000

### Request/Failure description

Both ergoCub hands do not calibrate.

### Detailed context

The logger is complaining about a CAN board not transmitting data. The readings from motorgui appear constant in most fingers

AntonioConsilvio, 2023-06-15:
Hi @S-Dafarra, yesterday I found the cable of the FAP (J31) of the thumb proximal detached. 

Then, I noticed that some open-closed joints fail to calibrate, so I reduced the final destination point that the fingers must reach to complete the calibration.

There will be a PR shortly.

cc @sgiraz @MSECode

AntonioConsilvio, 2023-06-27:
PR here

- https://github.com/robotology/robots-configuration/pull/532

closed 2023-06-27 by AntonioConsilvio